f4pga/prjxray #11: Document the bitstream for the IOBs in the Artix-7

mithro, 2017-12-22:
The IO blocks in the Artix-7 are quite complex and need to be documented.

tmichalak, 2019-01-22:
I still can't figure out the SLEW and DRIVE tags. For SLEW Vivado only allows to set the property to SLOW and FAST whereas for DRIVE the only ones allowed are 4,8,12 and 16. In both cases segmatch fails to resolve any candidates. I built on the design and entire flow for PULLTYPE, hence I used the add_site_group_zero function with all possible variations of the zero value tag.
I am not sure if the problem still lies in an incorrectly chosen zero value - instead of choosing an enum known to be zero I should simply choose a site which doesn't contain the enum, or whether I shouldn't use the add_site_group_zero function for adding the tags in the first place.

mithro, 2019-01-22:
@tmichalak It could be that SLEW + DRIVE don't affect the bitstream?

mithro, 2019-01-22:
@JohnDMcMaster - Can you help here?

mcmasterg, 2019-01-22:
Can you clarify? Is the issue that you aren't confident you know how the encoding works or that you aren't sure how to apply add_site_group_zero?

Re: encoding: have you diffed a few bitstreams to feel you are confident with the encoding? I recall I did some quick tests that showed that SLEW and DRIVE may not be two independent attributes. That is, you could individually change SLEW or DRIVE to come up with a model, but once you changed SLEW and DRIVE that model fell apart.

If the second, I'm not opposed to adding a "fake" segment that forces an enum to zero for the solver. This would simplify add_site_group_zero in that you would no longer have to provide an explicit zero entry.

tmichalak, 2019-01-23:
I think I understand the encoding for the PULLTYPE property, but I am not sure if the DRIVE and SLEW follow a similar pattern. Adding site tags without specifying a zero value actually yields results, but few bits overlap. Of course, I assumed that there is a separate model for SLEW and DRIVE, so the thing you said about those attributes not being independent is a bit worrying. 
Therefore I guess I will focus on adding a "fake" segment first. Any hints on this topic are welcome.

tmichalak, 2019-01-23:
After increasing the number of outputs in the design, since the DRIVE property can be set only on those, I started to get candidates with one of the properties as the zero value.
Another thing I tried is changing the IOSTANDARD to LVCMOS18 as it's possible to set more SLEW rates on the outputs, and used each of them as the zero value. The results were inconclusive though.
I would really need more info on the "fake" segment.

JohnDMcMaster, 2019-01-23:
It sounds like you are trying to add "zero" values without having a theory about how the bits are changing. I highly suggest you do a series of minitests and diff individual bitstreams with small changes before trying to make an IOB fuzzer.

Suggestion: Start with a base bitstream and create a series of bitstreams that cycle through all of the different SLEW and DRIVE properties so that you see the impact of each change individually. Then do the same exercise again, but now changing SLEW and DRIVE at the same time. Make a theory about how they are represented in the part and post it here for feedback. If all of that checks out, then go back to the fuzzer.

tmichalak, 2019-01-24:
Diffing small bitstreams is exactly the approach I have taken for the DRIVE property. I do have a list of differences, but I don't really see a pattern here (like in case of PULLTYPE). 
On top of that the differences tend to change a little bit with different iostandard, but are stable for LVCMOS18 for FAST and SLOW SLEW.

JohnDMcMaster, 2019-01-25:
Sorry super busy right now. Let me add a few quick comments / brain dump

Maybe it was the technology (ie LVCMOS18) not SLEW that I saw was changing other bits. I also recall seeing things like there was one property for an IOB diff pair, but it looked like there was actually one per IOB. My brief look at some of the drive strength attributes indicated it wasn't a trivial mapping between the property and the bits. With all of this, I'm thinking the following.

There are probably some low level configuration bits for the specific drive strength and such that could be even a large number of bits. They may have qualified the IOBs under certain conditions they know work well, and these are the officially supported settings. However, there may be a much larger number of knobs under the hood. My next proposal was to do a more proper automatic sweep of the different configs. Did you see the tcl minitest I created to do this? Take a look at the result, and hopefully that should give you some better data. Link: https://github.com/SymbiFlow/prjxray/blob/master/fuzzers/030-iob/minitest/sweep.tcl

tmichalak, 2019-01-25:
I created a PR with the minitest I used when looking for the changes in the bitstreams.
Looking at the differences I could observe the following:
1. The DRIVE settings are configured by mutliple bits
2. The SLEW doesn't influence the bits used for DRIVE
3. The bits that are different between SLOW and FAST SLEW are neither influenced by DRIVE or IOSTANDARD, i.e it's always the same bits that are different
4. For LVCMOS18 the bitstreams for DRIVE 8 and 12 are identical, for LVCMOS33 they differ by one bit

tmichalak, 2019-01-29:
In a short discussion with @JohnDMcMaster , he suggested to rename the tags to include the iostandard as well.
Apart from that we came to the conclusion that we need to automate the diffing process and generate the results for as many data as possible.
As part of the automation I wrote a python script that generates a csv file with the differences in the bits for various iostandards, slew and drive rate.
Now I have to update the script used for data generation to include all other formats besides CMOS18 and CMOS33.
The script has been pushed to the branch references in PR #564

tmichalak, 2019-02-04:
I added lvcmos12, lvcmos15, lvcmos25 and lvttl io standards with suitable drive strength values to the minitest's source data. I ran the csv compare and export tool I have written to browse the differences in a gui. The observations made several days ago still stand. Converting these result into a fuzzer isn't obvious though mostly due to the multiple bits nature of the configuration where single bits are present in more than just one feature. Is this even an option for that in the database?

litghost, 2019-02-04:
Is question whether a feature can have move than on bit?  Absolutely.

tmichalak, 2019-06-16:
We have a fuzzer that documents the bitstream for IOBs for quite some time now. Closing.

closed 2019-06-16 by tmichalak