f4pga/prjxray #12: Document the bitstream for the BRAM blocks in Artix-7

mithro, 2017-12-22:
The blockram parts are the last part needed to be documented to able to do a useful CPU design on the Artix-7.

mcmasterg, 2017-12-22:
I've started on this and made significant progress getting tilegrid and INIT fuzzer setup. More to come

mcmasterg, 2018-10-23:
just ran the full fuzzer for the first time and...there is an interleave issue. Only have of the bits fell out of the fuzzer. Probably related to BRAM18 vs BRAM36...will look into it tomorrow

JohnDMcMaster, 2018-11-12:
Yes. Basically:
- All of the bits that are needed for configuring core BRAM modes (ie excluding FIFO) should be decoded. This includes both bitstream sections
- Multibit issues should be fixed now (per mergedb changes last week)
- Interconnect fuzzer includes BRAM, but doesn't include all paths. I think it just needs a small tweak to instantiate additional primitives, but I need to poke at it

litghost, 2019-01-31:
The following bits are toggled by use of BRAM's, but are unknown bits:

```
# In frame 0x00400f1b 30 bits were not converted.
{ unknown_bit = "00400f1b_0_8" }
{ unknown_bit = "00400f1b_0_11" }
{ unknown_bit = "00400f1b_0_12" }
{ unknown_bit = "00400f1b_0_13" }
{ unknown_bit = "00400f1b_0_16" }
{ unknown_bit = "00400f1b_0_19" }
{ unknown_bit = "00400f1b_0_20" }
{ unknown_bit = "00400f1b_0_21" }
{ unknown_bit = "00400f1b_0_24" }
{ unknown_bit = "00400f1b_0_27" }
{ unknown_bit = "00400f1b_0_28" }
{ unknown_bit = "00400f1b_0_29" }
{ unknown_bit = "00400f1b_1_0" }
{ unknown_bit = "00400f1b_3_3" }
{ unknown_bit = "00400f1b_3_4" }
{ unknown_bit = "00400f1b_6_28" }
{ unknown_bit = "00400f1b_6_29" }
{ unknown_bit = "00400f1b_9_0" }
{ unknown_bit = "00400f1b_9_3" }
{ unknown_bit = "00400f1b_9_4" }
{ unknown_bit = "00400f1b_9_5" }
{ unknown_bit = "00400f1b_9_8" }
{ unknown_bit = "00400f1b_9_11" }
{ unknown_bit = "00400f1b_9_12" }
{ unknown_bit = "00400f1b_9_13" }
{ unknown_bit = "00400f1b_9_16" }
{ unknown_bit = "00400f1b_9_19" }
{ unknown_bit = "00400f1b_9_20" }
{ unknown_bit = "00400f1b_9_21" }
{ unknown_bit = "00400f1b_9_24" }
```

When only the Y0 BRAM18 is used the following bits are set:

```
# In frame 0x00400f1b 28 bits were not converted.
{ unknown_bit = "00400f1b_0_8" }
{ unknown_bit = "00400f1b_0_11" }
{ unknown_bit = "00400f1b_0_12" }
{ unknown_bit = "00400f1b_0_13" }
{ unknown_bit = "00400f1b_0_16" }
{ unknown_bit = "00400f1b_0_19" }
{ unknown_bit = "00400f1b_0_20" }
{ unknown_bit = "00400f1b_0_21" }
{ unknown_bit = "00400f1b_0_24" }
{ unknown_bit = "00400f1b_0_27" }
{ unknown_bit = "00400f1b_0_28" }
{ unknown_bit = "00400f1b_0_29" }
{ unknown_bit = "00400f1b_1_0" }
{ unknown_bit = "00400f1b_3_3" }
{ unknown_bit = "00400f1b_3_4" }
{ unknown_bit = "00400f1b_9_0" }
{ unknown_bit = "00400f1b_9_3" }
{ unknown_bit = "00400f1b_9_4" }
{ unknown_bit = "00400f1b_9_5" }
{ unknown_bit = "00400f1b_9_8" }
{ unknown_bit = "00400f1b_9_11" }
{ unknown_bit = "00400f1b_9_12" }
{ unknown_bit = "00400f1b_9_13" }
{ unknown_bit = "00400f1b_9_16" }
{ unknown_bit = "00400f1b_9_19" }
{ unknown_bit = "00400f1b_9_20" }
{ unknown_bit = "00400f1b_9_21" }
{ unknown_bit = "00400f1b_9_24" }
```
When only the Y1 BRAM18 is used the following bits are set:

```
# In frame 0x00400f1b 2 bits were not converted.
{ unknown_bit = "00400f1b_6_28" }
{ unknown_bit = "00400f1b_6_29" }
```

When both Y0 and Y1 BRAM18 are used, both sets of bits are set.  When the BRAM is used in BRAM36, both sets of bits are set.

The bits appear to follow which BRAM is used, so they do appear to be related to the BRAM is some way.

It is possible they are BRAM interconnect bits (e.g. #477), but based on results from #581.

JohnDMcMaster, 2019-01-31:
Convert these bits to segbits form and then look for them in the config fuzzer maybe? This should give you a lot more data on where they are used

litghost, 2019-02-01:
After solving the FIFO almost empty and almost full configuration, the follow bits are left:

RAMB36
```
# In frame 0x00400f1b 4 bits were not converted.                                  
{ unknown_bit = "00400f1b_3_3", unknown_segment = "0x00400f00", unknown_segbit = "27_99" }
{ unknown_bit = "00400f1b_3_4", unknown_segment = "0x00400f00", unknown_segbit = "27_100" }
{ unknown_bit = "00400f1b_6_28", unknown_segment = "0x00400f00", unknown_segbit = "27_220" }
{ unknown_bit = "00400f1b_6_29", unknown_segment = "0x00400f00", unknown_segbit = "27_221" }
```

RAMB18 Y0
```
# In frame 0x00400f1b 2 bits were not converted.                               
{ unknown_bit = "00400f1b_3_3", unknown_segment = "0x00400f00", unknown_segbit = "27_99" }
{ unknown_bit = "00400f1b_3_4", unknown_segment = "0x00400f00", unknown_segbit = "27_100" }
```

RAMB18 Y1
```
# In frame 0x00400f1b 2 bits were not converted.                               
{ unknown_bit = "00400f1b_6_28", unknown_segment = "0x00400f00", unknown_segbit = "27_220" }
{ unknown_bit = "00400f1b_6_29", unknown_segment = "0x00400f00", unknown_segbit = "27_221" }
```

mithro, 2019-02-03:
That is awesome!

Definitely looks like the 18 Y0 / Y1 bits and the 36 bits are related.

PR #608 (2019-02-05): Add fuzzer for BRAM/FIFO enable bits.
Fixes #12
closed 2019-02-05 by litghost