f4pga/prjxray #9: Convert README.txt to README.md files

mithro, 2017-12-22:
Markdown is formatted nicer on GitHub.

Lets fix up all our README files to be markdown.

mcmasterg, 2017-12-22:
All of them? I'm not a fan of this. I primarily view the repo for local use, not github. In that case they are nicer as .txt files

mithro, 2018-02-14:
Thanks to @unixb0y , the conversion has been done, but now they need a bit of clean up on the formatting. I think they should be formatted this way;

<pre lang="no-highlight"><code>
# CLBnFFMUX Fuzzer

## Purpose

Document nFFMUX family of CLB muxes  

## Algorithm

## Outcome

```
CLB.SLICE_X0.AFFMUX.B0 30_00
CLB.SLICE_X0.AFFMUX.B0 30_00
CLB.SLICE_X0.AFFMUX.B0 30_00
... more lines ...
```

From manual O6 testing ;

```
X0 AOUTMUX O6  
X0 BOUTMUX O6  
```

</code></pre>

which will then render like this;

> # CLBnFFMUX Fuzzer
> ## Purpose
> 
> Document nFFMUX family of CLB muxes  
> 
> ## Algorithm
> 
> ## Outcome
> 
> ```
> CLB.SLICE_X0.AFFMUX.B0 30_00
> CLB.SLICE_X0.AFFMUX.B0 30_00
> CLB.SLICE_X0.AFFMUX.B0 30_00
> ... more lines ...
> ```
> 
> From manual O6 testing ;
> 
> ```
> X0 AOUTMUX O6  
> X0 BOUTMUX O6  
> ```

unixb0y, 2018-02-14:
Ok @mithro if you want I’ll do that the next days. Didn’t know that was needed but it’s not a problem!

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

unixb0y, 2018-02-18:
Btw, this can be closed now.

mithro, 2018-02-18:
I think the bullet points still need to be fixed.

unixb0y, 2018-02-18:
Oh right, I remember.

mithro, 2018-02-18:
Btw @unixb0y are you student at Uni?

unixb0y, 2018-02-18:
Yeah I am 😃

unixb0y, 2018-02-18:
Pull Request is open 😊

mithro, 2018-02-18:
It is likely that both the [TimVideos](https://code.timvideos.us/summer-of-code/) and [FOSSi Foundation](https://fossi-foundation.org/gsoc18-ideas.html) will definately consider projects related to SymbiFlow for [Google Summer of Code](https://summerofcode.withgoogle.com/).

unixb0y, 2018-02-18:
Oh, thanks for the hint! I've heard of "Google Summer of Code" but didn't think of participating myself, especially with this project.
Might have a look at it after my exams 😉

closed 2018-02-18 by mithro