f4pga/prjxray #27: Getting rid of "segments" in tilegrid

cliffordwolf, 2018-01-01:
I think we should get rid of "segments" in tileconn.json. Instead we should have frame address information directly in the tile description. This would for example allow for INT tiles to have a height of two words, and BRAM/DSP tiles to have a height of 10 words, while referring to the same frame base address.

If we touch every script that uses tilegrid.json anyways this would also allow us to do some other cleanups. I would suggest storing the address information in individual attributes in a tile object, for example:

                "BRAM_L_X6Y55": {
                        "baseframe": "0x00400300",
                        "grid_x": 19,
                        "grid_y": 150,
                        "numframes": 28,
                        "numwords": 10,
                        "sites": {
                                "RAMB18_X0Y22": "FIFO18E1",
                                "RAMB18_X0Y23": "RAMB18E1",
                                "RAMB36_X0Y11": "RAMBFIFO36E1"
                        },
                        "type": "BRAM_L",
                        "wordoffset": 10
                },

I would make all necessary changes to the fuzzers, but we should first make sure we are all on the same page so we don't need to touch everything twice. So please make sure to leave your comments below.

mithro, 2018-01-01:
It is likely that I will end up having segments in vpr (as it prefers having the interconnect and logic in the same tile).

cliffordwolf, 2018-01-01:
> It is likely that I will end up having segments in vpr

How are you going to deal with BRAMs and DSPs in this case? The current segment system splits them each into 5 smaller "pseudo tiles". In case of DSPs you then have 2 DSP slices distributed over 5 such pseudo tiles, i.e. 2.5 tiles for each DSP slice. I don't know VPR well but I don't think this would make it very happy.

mithro, 2018-01-01:
vpr has tiles which spans multiple grid locations.

cliffordwolf, 2018-01-01:
> vpr has tiles which spans multiple grid locations.

I know. But (1) that doesn't help you if you want to merge BRAM + INT, because then they do not have the same size anymore. (INT is still only one unit / two words in height. That's the whole reason I want to get rid of the segments construct, so that this causes no problems.) The only thing you could do is create a tile that combines one BRAM and 5 INT tiles, but there is nothing to gain by using this representation in prjxray. It would just make things more complicated because we couldn't reuse the INT information anymore the way we do now. And (2) then you would use a different segmentation than the tilegrid "segments", so it wouldn't help you in any way if we keep the segments in the tilegrid.

kc8apf, 2018-01-01:
I've been thinking about this from the generating config frames side as
well.  I want to stop using 32-bit frame addresses in our config files.
That representation loses a lot of semantic details that will become
important such as when you are crossing a clock domain.  I expect the
per-part YAML files to be extended to describe the row-half, row, column
structure.  Tiles would then be described as a tile type, specific column
(row-half, row, column address), and a 2D area (minor frame begin/end, word
begin/end).  Tile types would be defined in a shared definition as they are
consistent throughout the entire part series.  There may be reason to
introduce column types as only a few column types are repeated throughout
the chip.

cliffordwolf, 2018-01-04:
I'm gravitating towards packing frame base addr, num of frames, word offset and number of words into one tile property, and to add two such properties for BRAMs that have config bits in one part of the bit-stream and data bits in another one. The new definition of "segment" would be such a 4-tuple. For example:

                "BRAM_L_X6Y55": {
                        "configseg": "00400300:28:20:10",
                        "dataseg": "00c00000:128:20:10",
                        "grid_x": 19,
                        "grid_y": 150,
                        "sites": {
                                "RAMB18_X0Y22": "FIFO18E1",
                                "RAMB18_X0Y23": "RAMB18E1",
                                "RAMB36_X0Y11": "RAMBFIFO36E1"
                        },
                        "type": "BRAM_L",
                },

kc8apf, 2018-01-04:
I find the packed configseg and dataseg properties difficult to comprehend.  I need to go lookup the definition to figure out how to interpret it.  Part of my work in https://github.com/kc8apf/prjxray/tree/part_refactor is to describe the configuration space independent of tiles. A tile can reference a configuration region via a YAML alias such as:

- BRAM_L_X6Y55:
    configseg: 
      column: &that_particular_column
      offset: 20
      length: 10
    dataseg: 
      column: &other_column
      offset: 20
      length: 10
    grid_x: 19
    grid_y: 150
    ....
    
The referenced column already describes the base frame address and the number of frames:
!<xilinx/xc7series/part>
idcode: 0x362c093
global_clock_regions:
  top: !<xilinx/xc7series/global_clock_region>
    rows:
      0: !<xilinx/xc7series/row>
        configuration_buses:
          CLB_IO_CLK: !<xilinx/xc7series/configuration_bus>
            configuration_columns:
              0: !<xilinx/xc7series/configuration_column>
                frame_count: 42
              1: !<xilinx/xc7series/configuration_column>
                frame_count: 30
              2: !<xilinx/xc7series/configuration_column>
                frame_count: 36

cliffordwolf, 2018-01-06:
Finally got an hour to reply to this. I'm going to paraphrase your suggestion and speculate about motivations below. Please let me know if I misrepresent your position, as this is not my intention. I hope we can resolve this soon.

Your suggestion as I understand it:

- Store in tile in tilegrid: Ref. to column, word offset, word length
- Number of frames for the column can be resolved by looking at the referenced column in the part description and reading the "frame_count" property.
- Frame base address can be resolved by looking up the referenced column in the part description, and following the tree upwards to figure out the column number, the "configuration bus", the row, and if it is in the top or bottom half of the chip.
- You want to use YAML for everything

You presumed motivation:

- Deduplication of data
- You mentioned that you want the frame_count in the part description for processing bitstreams that use implicit FAR increments.
- I currently have no data why you prefer YAML.

My suggestion:

- Store in tile in tilegrid: Frame base address, number of frames, word offset, word length
- I prefer JSON over YAML for serialized data

My motivation:

- I want it to be as easy as possible to get the frame base address for a tile for things like segmaker.py.
- Right now the fuzzers only need tilegrid. I'd like to keep it that way for simplicity.
- I want it to be as easy as possible to correlate to segment names in segmaker output to actual tiles.
- JSON is easier to parse and in wider spread use as serialization format. Also e.g. the python "json" module is part of the standard lib, but PyYAML is not.

----

So here is the outline of a compromise that I'm proposing:

Afaics I don't actually need the frame_count in any of my tilegrid users. So if you want to store that in the parts description only then I'm fine with that.

I don't have a very strong opinion on YAML vs JSON, but I think YAML is better for (hand written) config files and JSON is better suited as data serialization format. I think I'd be fine with using YAML as file format, if there is a strong argument for it. But, and this is very important for me, I wouldn't want to use features like YAML references across files! (I did not even know that this is a thing. Also: Shouldn't it be `*other_column` in the tile grid and then `&other_column` in the part description?)

If there is a good argument for YAML over JSON then I'd like to hear it. But at the same time I think that JSON vs YAML is a red herring that's distracting us from the actual issues. I'd actually prefer to keep tilegrid and tileconn as JSON and the part description as YAML if it helps avoiding a discussion about the formats.

In your suggestion you introduce unique reference labels for each column (`that_particular_column`, `other_column`). How exactly would you construct those labels? Obviously you are not literally suggesting names like `that_particular_column` and `other_column`. The frame base address *is* unique for each column. So why not just use that!? Then we'd have the frame base addresses in tilegrid (like I want), and we have a simple link between tilegrid and column in part description (like you want).

I'd leave it up to you if you want to have an explicit per-column property for the frame base address in the part description, or want to decode the frame base address and then do the hierarchical lookup for the column in the parts description. But I'd like to note that with your suggestion you would have needed to attach a unique label to each column anyways, so adding an explicit frame base address property to a column in the parts description would not be so different from what you have suggested originally.

My preferred way of encoding `configseg` would actually be a single string, such as `<hex-frame-base>:<word-offset>:<word-length>` (or similar) instead of using an object with multiple properties. The main motivation here is that I'd be able to use the same string as seg name in the files generated by segmaker, which would dramatically simplify tasks like manually correlating those files with the tilegrid (because it's much easier to search for a single string than for an object with a certain set of properties when looking at the tilegrid db in a text editor).

So, in example code, this is my suggestion:

```
            "BRAM_L_X6Y55": {
                    "configseg": "00400300:20:10",
                    "dataseg": "00c00000:20:10",
                    "grid_x": 19,
                    "grid_y": 150,
                    "sites": {
                            "RAMB18_X0Y22": "FIFO18E1",
                            "RAMB18_X0Y23": "RAMB18E1",
                            "RAMB36_X0Y11": "RAMBFIFO36E1"
                    },
                    "type": "BRAM_L",
            },

!<xilinx/xc7series/part>
idcode: 0x362c093
global_clock_regions:
  top: !<xilinx/xc7series/global_clock_region>
    rows:
      0: !<xilinx/xc7series/row>
        configuration_buses:
          CLB_IO_CLK: !<xilinx/xc7series/configuration_bus>
            configuration_columns:
              0: !<xilinx/xc7series/configuration_column>
                frame_base: "00000000"
                frame_count: 42
              1: !<xilinx/xc7series/configuration_column>
                frame_base: "00000080"
                frame_count: 30
              2: !<xilinx/xc7series/configuration_column>
                frame_base: "00000100"
                frame_count: 36
```

kc8apf, 2018-01-08:
I'm making a few assumptions that lead to breaking down the frame base
address into its components:
- To read non-PERFRAMECRC bitstreams, a complete map of valid frame
addresses within a part is required
- The above is independent of any knowledge of tiles (i.e. it can be
generated before any tiles are identified)
- Tiles are logically grouped inside clock regions and knowing that
grouping is helpful
- Clock regions and global clock regions will eventually have information
that applies to the whole region and all tiles within them

I looked through and experimented with many JSON and YAML libraries for
C++.  For serializing/deserializing C++ types, I found yaml-cpp to offer
the best usability.  YAML, as a format, offers features such as tags the
aid in specifying intent that helps with decoding (i.e. this is supposed to
be this complex type).  Includes and references provide a supported
mechanism for separating per-part and per-series information into separate
files while being able to stitch them together easily at parse time.  This
can be done with JSON as well but requires replicating the mechanisms
provided by YAML's base spec.

I view the part definition as data that needs to be both human and machine
understandable.  I find explicit encoding of groupings and relationships in
the file to be easier to understand and remember than needing to memorize
them.  The downside is needing to decode frame addresses to look up
information by hand.  That's why I wrote the frame_address_decoder tool.

Regardless of how and where tiles are defined, I need a separate
description of the frame addressing for general bitstream decoding to
work.  I'm proposing that this description be leveraged by the tile
description to reduce duplication and errors.  I'm hearing that you would
like to continue using encoded frame addresses in tilegrid.  I'm also
hearing that you're not swayed my experiences and rationale for introducing
the part YAML.  At some point tilegrid and part YAML need to meet for a
full toolchain to work.  As I haven't seen an alternative proposed for the
part YAML, I'm going to keep going down the path I'm on.  If that
ultimately means generating JSON from the processed part YAML so existing
tools continue to work, that's what I'll do.

Regarding segmaker, segmatch, etc: my plan, based on what I understand to
be my role in the project, is to rewrite tools like segmatch to use these
C++ libraries and handle all this file format detail there.  Now that the
C++ libraries can decode bitstreams into frames, segmatch can be changed to
operate directly on a bitstream rather than require bitread to be run
first.  I haven't looked at segmaker to see what makes sense there.

cliffordwolf, 2018-01-09:
Even after reading your comment a few times I don't understand how what you are writing relates to the concrete suggestions I made in my comment. I don't know if the format I suggested is okay or not in your opinion, and I don't know if you think using the frame base address as link between tilegrid and pard description is okay. So I guess I'll just wait further and not touch anything.

> Regarding segmaker, segmatch, etc: my plan, based on what I understand to
be my role in the project, is to rewrite tools like segmatch to use these
C++ libraries and handle all this file format detail there.  Now that the
C++ libraries can decode bitstreams into frames, segmatch can be changed to
operate directly on a bitstream rather than require bitread to be run
first.  I haven't looked at segmaker to see what makes sense there.

So you are saying segmaker, which doesn't need the "part description" currently and is pretty simple and straight forward python module right now, will in the future need to use C++ libraries to assemble frame base addresses by combining the information from multiple files using a complicated scheme because you want to remove the frame base address from tilegrid? If so: Please do me a favor and at least provide a rationale why any of this is an improvement over my suggestion in your opinion. Because I can't see the advantage of it. (And to be clear: Explaining why you think it is an improvement over my suggestion requires referencing the thing I actually suggested!)

> I'm making a few assumptions that lead to breaking down the frame base
address into its components:
> - To read non-PERFRAMECRC bitstreams, a complete map of valid frame
addresses within a part is required
> - The above is independent of any knowledge of tiles (i.e. it can be
generated before any tiles are identified)
> - Tiles are logically grouped inside clock regions and knowing that
grouping is helpful
> - Clock regions and global clock regions will eventually have information
that applies to the whole region and all tiles within them

I don't see how any of this relates to the questions I raised, or any of that would be prevented by using the format I suggested.

This whole interaction is extremely frustrating for me because I don't understand how the things you bring up relate to the questions I raised, and I can't do any work without having the questions I raised clarified.

So for the time being I'm now giving up: I can't work on that without some kind of agreement on the formats, and I can't discuss the formats because the questions I raised are completely ignored and the replies I get have no obvious relation to the questions I need answered, and/or completely ignore the arguments I brought up. So I'm now going to focus my resources other stuff (not related to prjxray).

closed 2018-01-09 by cliffordwolf