f4pga/prjxray #46: Document MMCM config bits for Artix-7

azonenberg, 2018-01-12:


azonenberg, 2018-01-12:
Much of the config is accessible by the DRP so this might be as simple as figuring out what DRP addresses map to what bitstream addresses, plus maybe some interconnect

mcmasterg, 2018-01-17:
So far we've been doing everything via the fuzzer approach. Unless there is a reason why we think fuzzer route will be significantly more difficult, lets stick with that for consistency

kgugala, 2019-07-04:
@emka what is the status of this one? I see some commits related to this issue in your fork. Can you write a few words here what you already did?

emka, 2019-07-21:
I correlated the CLKOUT_DIVIDE bits, the first one is merged and I found the others to behave identical. After the last christmas break I made no further progress unfortunately.

mkurc-ant, 2020-10-27:
Status update: There are two fuzzers related to `MMCM` and the `CMT_TOP_[LR]_LOWER_B` tile:
- MMCM features: https://github.com/SymbiFlow/prjxray/tree/master/fuzzers/031-cmt-mmcm
- Tile PIPs: https://github.com/SymbiFlow/prjxray/tree/master/fuzzers/034b-cmt-mmcm-pips

These solve most of `MMCM` bits required to use it.

mkurc-ant, 2020-10-27:
I've created a separate issue to track documenting bits for advanced features of `MMCM`: https://github.com/SymbiFlow/prjxray/issues/1473

tmichalak, 2021-01-21:
Superseded by #1473.

closed 2021-01-21 by tmichalak