f4pga/prjxray #56: Tie python tests into CIT

mcmasterg, 2018-01-22:
There are some test_* things in tools/ that should ideally be run as part of submit check

mithro, 2019-04-04:
@elmsfu - Is this done now?

elmsfu, 2019-04-04:
> @elmsfu - Is this done now?

Not to my knowledge SymbiFlow/symbiflow-arch-defs#520 runs tests under symbiflow-arch-defs/utils.

We should consider gathering coverage information.

mithro, 2019-04-05:
Hrm, the Python tests do seem to be running?
```
if [ -e env/bin/activate ]; then . env/bin/activate; fi; which py.test; py.test --ignore third_party --ignore .git --ignore env --ignore build --ignore docs/venv --ignore fuzzers --ignore minitests --ignore experiments --doctest-modules --junitxml=build/py_test_results.xml
/home/travis/build/SymbiFlow/prjxray/env/bin/py.test
============================= test session starts ==============================
platform linux -- Python 3.6.3, pytest-4.4.0, py-1.8.0, pluggy-0.9.0
rootdir: /home/travis/build/SymbiFlow/prjxray
collected 15 items                                                             
docs/markdown_code_symlinks.py .                                         [  6%]
prjxray/lib.py .                                                         [ 13%]
prjxray/util.py .                                                        [ 20%]
tests/test_util.py .                                                     [ 26%]
utils/cmp.py .                                                           [ 33%]
utils/sort_db.py .........                                               [ 93%]
utils/xjson.py .                                                         [100%]
```

elmsfu, 2019-04-05:
Closing. Looks like you fixed it in #491

closed 2019-04-05 by elmsfu