f4pga/prjxray #57: "FPGA assembly": naming

mcmasterg, 2018-01-22:
"FPGA assembly" (.fasm file) as its currently called, refers to an end user issuing low level instructions in a human readable text format to configure FPGA interconnect and logic elements such as FFs and LUTs.

However, we are unclear if this is the best name. Given that we expect this to be a semi core concept in the tools, we'd like to get a reasonable term to describe it early on. This ticket is about trying to make a decision on record about that choice.

@kc8apf asked on twitter for some feedback. Not sure if ##openfpga was asked as well? I'll follow up with some of the name suggestions and my quick thoughts on them

Note semi related ticket "FPGA assembly: syntax" is being created

kc8apf, 2018-01-23:
Some of the twitter responses:

@mithro: ax7s
@SamuelAFalvoll: GACL - Gate Array Construction Language
@KWF: RHDL - Register Hardware Description Language
@OlofKindgren: Reggae - Register Gate Array Expressions
@natsfr: FFlopDL - Flip Flop Description Language

I like GACL the best so far.

mcmasterg, 2018-01-23:
GACL is the favorite of those so far for me

mithro, 2018-12-07:
We have decided to stick with `fasm`.

closed 2018-12-07 by mithro