f4pga/prjxray #58: "FPGA assembly": syntax

mcmasterg, 2018-01-23:
"FPGA assembly" (.fasm file) as its currently called, refers to an end user issuing low level instructions in a human readable text format to configure FPGA interconnect and logic elements such as FFs and LUTs. This could be hand written our dumped by a PNR tool. As an FPGA user should not generally be aware of segments or other internal bitstream details, this syntax was specifically created to not have any knowledge of segments or frames.

There have been some semi heated discussions about what syntax we should use. Today, we have a format that is very explicit but perhaps too verbose. There have been some suggestions to simplify it to match segprint closer, or to eliminate segments entirely and unify the two formats. I'm going to try to describe some of the background on this and then follow up with some more personal opinions.

Issue 27 describes proposals to change tilegrid: https://github.com/SymbiFlow/prjxray/issues/27 Now that I have more background on the bitstream format, I need to review this in more detail to better understand what the core issues are.

There was also some discussion on twitter: https://twitter.com/oe1cxw/status/954402448436756480

This was followed up with a brief chat with Clifford where he encouraged us to reference issue 27 regarding his thoughts on this matter. This implies to me that he considers this ticket and issue 27 the same thing (that is if we eliminate tiles segprint will produce something usable to end users).

So basically the questions are:
-How closely related should the format used to document the FPGAs be to the format our PNR tools spit out?
-Is tile based naming the right level for this?
-How closely do we want to follow Vivado naming?
-How verbose should we be? For example, some are advocating for implicit elements to be removed from the syntax.

mcmasterg, 2018-01-23:
From some quick discussions, it sounds like I should go ahead and change the way that I name slices to use relative names instead of global names. The reason why its like this right now is because originally I wanted names that look like this:

SLICE_X12Y102.AFF.DMUX.AX 1

However, Rick convinced me that to start we should start everything with a tile name to be more regular. I conceded to this with the comment that I might want something like above supported as an extension if we do a lot of FPGA assembly operations. So I did the most straightforward thing which was to simply prepend the tile name:

CLBLM_L_X10Y102.SLICE_X12Y102.AFF.DMUX.AX 1

But now this has two global coordinates in it. We probably want something with X0/X1 to make it relative to the tile. If we use segprint convention it might look something like this:

CLBLM_L_X10Y102.SLICEM_X0.AFF.DMUX.AX 1

I'll probably default to this for now as it seems the most regular vs what we already have named. Aside from that, I have mixed feelings about naming "SLICEM_X0" and "SLICEL_X1". That is, prefixing with SLICEM or SLICEL indicates to me you are enumerating elements of that type. However, in this case there is only one SLICEL, therefore it should be SLICEL_X0. Of course, this is done to match the slice X0/X1 notation, but then I'd argue you might just be better off omitting the slice type:

CLBLM_L_X10Y102.X0.AFF.DMUX.AX 1

Where the user would just get a compile error if they tried to use something that doesn't apply in SLICEL. The previous notation is of course more explicit, but I'm not sure it buys much here.

mcmasterg, 2018-01-23:
Another point of contention is how the value syntax should be supplied. segprint make the  value implicit by doing things like:

tag CLBLM_L.SLICEM_X0.AFF.DMUX.AX

Where I was suggesting to have an explicit value argument:

CLBLM_L_X10Y102.SLICE_X12Y102.AFF.DMUX.AX 1

At least to me, this makes a lot more intuitive sense in a key/value sense, especially when conveying routing information. For example, segprint type syntax would loosely look like:

INT_L_X10Y102.CLK_L1.GCLK_L_B11_WEST

Which is interpreted as INT_L_X10Y102.CLK_L1 has value GCLK_L_B11_WEST. However, I find this a lot less intuitive given that we often use . separators to represent hierarchy, not a configuration value.

For example, if I saw these in a file:

INT_L_X10Y102.CLK_L1.GCLK_L_B11_WEST
INT_L_X10Y102.CLK_L1.GCLK_L_B5

Its not immediately obvious they are conflicting directives. But if I saw:

INT_L_X10Y102.CLK_L1 GCLK_L_B11_WEST
INT_L_X10Y102.CLK_L1 GCLK_L_B5

I'd immediately know those were both INT_L_X10Y102.CLK_L1 directives, and cannot be used together.

With all that in mind, I have no objections to making an omitted value for a single bit configuration equivalent to supplying the value 1. That is, the following two directives would be equivalent:

CLBLM_L_X10Y102.SLICE_X12Y102.AFF.DMUX.AX 1
CLBLM_L_X10Y102.SLICE_X12Y102.AFF.DMUX.AX

But this would be a syntax error:

INT_L_X10Y102.CLK_L1

cliffordwolf, 2018-01-23:
Putting in my two cents:

> INT_L_X10Y102.CLK_L1.GCLK_L_B11_WEST  
Which is interpreted as INT_L_X10Y102.CLK_L1 has value GCLK_L_B11_WEST.

That is not the intended semantics of the segbits tag syntax. It just means that the pattern `INT_L_X10Y102.CLK_L1.GCLK_L_B11_WEST` is active. There is no "key" and "value" part.

The reason I think `INT_L_X10Y102.CLK_L1.GCLK_L_B11_WEST` is better than `INT_L_X10Y102.CLK_L1 GCLK_L_B11_WEST` is because a tool working with this file format does not need to know that this pattern is a PIP in order to process or generate the statement. It is sufficient to know that there is a pattern with this name. With the suggested format a tool working with the files would needs to know which pattern are PIPs and which aren't in order to know when to replace the last dot with a blank.

Furthermore, there are PIP-like pattern for some (but not all) routing primitives. For example `CLBLM_L.SLICEL_X1.AFF.DMUX.O6` vs. `CLBLM_L.SLICEL_X1.AFF.DMUX O6`. So either the file format is inconsistent by using the dot syntax for those, or a simple heuristic like "when it's a pattern in an INT tile" is insufficient to figure out when the last dot needs to be replaced by a blank.

Similarily here: `CLBLM_L_X10Y102.SLICE_X12Y102.AFF.DMUX.AX`. With this syntax a tool must know that `SLICE_X12Y102` maps to `SLICEM_X0` in order to look up the bit pattern. When writing a file it's even more overhead because there is no algorithm for calculating slice positions from tile positions, so one would always need to look up the name of the site in tilegrid to be able to generate the tag name.

What possible advantage (beyond a slight aesthetic difference) could that provide? I honestly don't see any, but if there is one I'd be eager to learn about it. All I can see at the moment is that the suggested format makes it significantly harder to write tools that read and write with those files.

I think the syntax should just be 1:1 the name as we use it in the database just with the first part (the tile type) replace with the tile name. I don't see why any changes to the tag names other than that would be needed. Afaict it would only come with the disadvantage that it is (a) confusing for someone new approaching the project because the names are not consistent in the different files and (b) each software that reads or writes those files and uses the segbits database in any way must be aware of those rewrite patterns and implement them in order to correlate the entries in the two files. When we add new things (like bits for DSPs and/or BRAMs) we would need to go back to all those tools and add the missing rules for the new tags that have been added. None of this would be necessary if we just use a consistent naming scheme without any special tweaks.

Note that I don't say the tag names in the database are perfect! I'm just saying they should be consistent between the database and the fasm file. If a name in the database is counter-intuitive or otherwise bad then the pattern should be given a different name in the database, not just given a different name in the fasm file and each software working with those files burdened by having to translate between the different names.

mcmasterg, 2018-01-23:
TLDR:
A) I'm fine with matching the existing names, except that I will prefix them with tiles instead of segments
B) If there are any issues with naming, we'll propose changing them in tilegrid. This would also encompass things like the PIP junction box naming if we ever want to revisit that
C) My main interest is moving things forward good enough. I do agree that this is probably the best solution with that in mind

Rick/Clifford: can you both give me a "SGTM" and I'll consider this resolved?

Details:

1) Intended semantics: I think you are favoring making tools easier to write, while I'm thinking what I would want as a user. If we think that this is so low level that its unreasonable to think users will use it in the long run (which I think I could agree with) then I would agree that favoring easier tools is more important than usability

2)  PIP like non routing: these are single bits. I find that to be more relevant than the fact that they happen to be routing. You are thinking of things in terms of what something does, I'm thinking in terms of how something is configured.

3) "CLBLM_L_X10Y102.SLICE_X12Y102" This was already agreed to be undesirable and there is already a PR to change this

4) Aesthetics: I think aesthetics are important, although see previous comment that I"m agreeing it may not matter in favor of easier to write tools

5) I would agree that consistent naming is important where possible

6) I would agree that some of the comments are now possibly more related to database tag names. This confusion was mostly caused because I was proposing to originally name slices based on the slice, and now we've agreed they should be named relative per tile.

cliffordwolf, 2018-01-23:
First: SGTM

Just two small comments regarding 1. and 2.:

> 1. Intended semantics: I think you are favoring making tools easier to write, while I'm thinking what I would want as a user.

I think I have both in mind. As a user, I also want to be able to for example simply `grep` for a pattern in the database files. This is complicated (and error prone) if I first have to apply a non-trivial transformation to the pattern.

> 2. PIP like non routing: these are single bits.

No, they are not. For example `CLBLM_R.SLICEL_X1.CFF.DMUX.*` uses two bits in 50% of the patterns:

[image]

mcmasterg, 2018-01-23:
2) ah yes I am incorrect there. Still, I would argue that is more intuitively a key value

JohnDMcMaster, 2018-11-16:
@litghost made a push on this recently

mithro, 2018-12-07:
@litghost Can you update this issue?

litghost, 2018-12-19:
@mithro Sure

https://github.com/SymbiFlow/fasm has the start of a library for parsing and emitting FASM files from python.  There is a Google doc that tries to formalize the meaning of FASM that I need to convert to a .md and add to the repository, but it generally matches the conversation above.  I believe there also needs to be a C-linkable FASM parsing library, but that isn't a priority at this time.  Something that doesn't exist, but should be done is a "FASM style guide", that attempts to guide database designers to choose good names and hierarchies.

I've created prjxray specific fasm tooling in the form of:

- https://github.com/SymbiFlow/prjxray/blob/master/utils/fasm2frames.py - Converts FASM input to frames suitable to be used with xc7patch.  This is how symbiflow-arch-defs is emitting bitstreams
- https://github.com/SymbiFlow/prjxray/blob/master/utils/fasm_pprint.py - The beginning of a FASM pretty printer for 7-series FASM files.  Right now it is usefully to generate the "canonical" form of a FASM file, which should allow diffing of FASM files.

I've added FASM support to VPR in the form of metadata tags, and are using these metadata tags to emit FASM from VPR place and routed design.