f4pga/prjxray #73: fuzzers/054-gfan occasionally fails due to DRC errors

kc8apf, 2018-02-02:
As I setup the continuous integration system, the jobs occasionally fail on 054-gfan due to a DRC error:

[91mERROR: [DRC RTSTAT-5] Partial antennas: 1 net(s) have a partial antenna. The problem bus(es) and/or net(s) are mynet_5.

I'm seeing if I can grab the workspace to extract the exact specimen that is triggering this.

kc8apf, 2018-02-02:
Container Builder doesn't preserve the workspace.  I'll try running the fuzzer a bunch on my workstation and see what happens.

kc8apf, 2018-02-08:
I managed to reproduce locally.  Tarball of the fuzzer after failure is attached.  specimen_005 failed with an antenna on mynet_1.  Vivado log was rotated when I opened the .dcp in Vivado so look at vivado_63010.backup.log for the output from the fuzzer run.
[gfan-antenna.tar.gz](https://github.com/SymbiFlow/prjxray/files/1705341/gfan-antenna.tar.gz)

mithro, 2018-02-09:
Is this related to #77 ?

kc8apf, 2018-02-09:
Unclear.

mithro, 2018-02-23:
Just had another failure of `054-gfan` fuzzer but I can't find the actual errors that caused it. 

I just see;
```
make[5]: Leaving directory '/source/fuzzers/054-gfan'
Makefile:24: recipe for target 'run' failed
make[4]: Leaving directory '/source/fuzzers/054-gfan'
[91mmake[4]: *** [run] Error 2
[0mMakefile:25: recipe for target '054-gfan/run.ok' failed
[91mmake[3]: *** [054-gfan/run.ok] Error 2
make[3]: *** Waiting for unfinished jobs....
```

mithro, 2018-02-23:
I'm adding "--output-sync=target" in https://github.com/SymbiFlow/prjxray/pull/102 to make it easier to see what the output looks like.

mithro, 2018-12-07:
@JohnDMcMaster I believe you fixed this recently?

mcmasterg, 2018-12-10:
Its improved, although not sure its fully fixed. Specifically I fixed antenna net issues (which were most of them?), but IIRC there was another class of error that could still come up

closed 2019-01-17 by litghost