f4pga/prjxray #74: Rename the slices inside CLBs

mithro, 2018-02-02:
Currently we use the following names in the database for the slices inside CLBs;

 * `CLBLL_(R|L)`
    * `SLICEL_X0` - First SLICE_L
    * `SLICEL_X1` - Second SLICE_L
 * `CLBLM_(R|L)`
    * `SLICEL_M0` - The SLICE_M
    * `SLICEL_X1` - The SLICE_L

I'm unsure of were these name originated from but I think they should instead be labelled the following way;

 * `CLBLL_(R|L)`
    * `CLBLL_L` - First SLICE_L
    * `CLBLL_LL` - Second SLICE_L
 * `CLBLM_(R|L)`
    * `CLBLM_M` - The SLICE_M
    * `CLBLM_M` - The SLICE_L

This is to be more consistent with the names given inside the pseudo pip database files (`ppips_clblX_X.db`).

For example, see the lines below from [`ppips_clbll_l.db`](https://github.com/SymbiFlow/prjxray-db/blob/master/artix7/ppips_clbll_l.db);
```
CLBLL_L.CLBLL_L_AX.CLBLL_BYP0 always
CLBLL_L.CLBLL_LL_AX.CLBLL_BYP1 always
CLBLL_L.CLBLL_L_CX.CLBLL_BYP2 always
CLBLL_L.CLBLL_LL_CX.CLBLL_BYP3 always
CLBLL_L.CLBLL_LL_BX.CLBLL_BYP4 always
```

Using the names found in this file I know that the CLBLL's `BYP0` pin is always connected to the `AX` pin of the *first* slice L inside the CLBLL -- IE `CLBLL.`**`L`**.  As well the CLBLL's `BYP1` pin is always connected to the `AX` pin of the *second* slice L inside the CLBLL -- IE `CLBLL.`**`LL`**.

And the example from [`ppips_clblm_l.db`](https://github.com/SymbiFlow/prjxray-db/blob/master/artix7/ppips_clblm_l.db);
```
CLBLM_L.CLBLM_L_AX.CLBLM_BYP0 always
CLBLM_L.CLBLM_M_AX.CLBLM_BYP1 always
CLBLM_L.CLBLM_L_CX.CLBLM_BYP2 always
CLBLM_L.CLBLM_M_CX.CLBLM_BYP3 always
CLBLM_L.CLBLM_M_BX.CLBLM_BYP4 always
CLBLM_L.CLBLM_L_BX.CLBLM_BYP5 always
```

Using the names found in this file I know that the CLBLM's `BYP0` pin is always connected to the `AX` pin of the SLICE_**L** inside the CLBLM -- IE `CLBLM.`**`L`**.  As well the CLBLM's `BYP1` pin is always connected to the `AX` pin of the SLICE_**M** inside it -- IE `CLBLM`.`**`M`**.

In theory I could try just mapping `SLICE_X0` to `CLBLL_L` or `CLBLM_M` and `SLICE_X1` to `CLBLL_LL` or `CLBLM_L` but it gets a bit confusing because the "`L` wires" map to the `SLICE_X0` site inside a CLBLL and to the `SLICE_X1` site inside the CLBLM.

This all comes from trying to interpret these names as references to actual objects which are connected. This attempt might be the start of naming scheme to be something like this; `CLBLM.L:BX->CLBLM:BYP5 always` which reads as the `BX` pin of the `L` object inside the `CLBLM` object is connected to the `BYP5` pin of the `CLBLM` object. This generalizes the names into `<OBJECT>`(`.<SUBOBJECT>`*)?`:<PIN>`

Another possibility might be - `CLBLM_Left.L:BX->CLBLM_Left:BYP5 always`

[An example of how the resulting database would change can be seen here.](https://github.com/SymbiFlow/prjxray-db/compare/master...mithro:rename) but it unclear to me how the fuzzers would have to change to generate this output.

----

An alternative would be to change the names in the pseudo pip files as follows;
```
CLBLM_L.SLICEL_X1_AX.CLBLM_L_BYP0 always
CLBLM_L.SLICEM_X0_AX.CLBLM_L_BYP1 always
```
and
```
CLBLL_L.SLICEL_X0_AX.CLBLL_L_BYP0 always
CLBLL_L.SLICEL_X1_AX.CLBLL_L_BYP1 always
```

mithro, 2018-02-02:
@mcmasterg -- As we were discussing after trying to get this data into VPR.

mithro, 2018-02-06:
@cliffordwolf Thoughts?

cliffordwolf, 2018-02-06:
First, allow me to correct two imo very relevant typos:

----

Currently we use the following names in the database for the slices inside CLBs;

 * `CLBLL_(R|L)`
    * `SLICEL_X0` - First SLICE_L
    * `SLICEL_X1` - Second SLICE_L
 * `CLBLM_(R|L)`
    * **`SLICEM_X0`** - The SLICE_M
    * `SLICEL_X1` - The SLICE_L

I'm unsure of were these name originated from but I think they should instead be labelled the following way;

 * `CLBLL_(R|L)`
    * `CLBLL_L` - First SLICE_L
    * `CLBLL_LL` - Second SLICE_L
 * `CLBLM_(R|L)`
    * `CLBLM_M` - The SLICE_M
    * **`CLBLM_L`** - The SLICE_L

----

I'm not fundamentally opposed to changing the site names to `CLBLL_L/CLBLL_LL` and `CLBLM_M/CLBLM_L`. But I think we should be absolutely clear about *what* we are doing here and *why*.

Re wire vs slice names: The wires (and PIPs) are in the tile outside the slices. The name of the slice is for example `SLICE_X9Y52` (which is the L slice inside the `CLBLM_R_X7Y52` tile in the kintex database). The other slice in that tile is the M slice `SLICE_X8Y52`. There is no object within the tile in the Xilinx database that is called  `CLBLL_L/CLBLL_LL` or `CLBLM_M/CLBLM_L`. It's just a prefix used for some of the wire names.

So I think `CLBLM_R.SLICEL_X1` and `CLBLM_R.SLICEM_X0` are quite reasonable names that match the Xilinx names very well.

In addition to that the proposed new names suggest imo that the sllices are CLBs (which they are not, the tile is a CLB and each CLB contains two slices), and that further weakens consistency between our naming scheme and the names Xilinx is using.

So the main question here is: Do we want a naming scheme that is close to the names Xilinx is using for those things (which would be the existing scheme), or do we want a naming scheme that diverges from the names Xilinx is using in favor for one that seems "more intuitive" to us.

> In theory I could try just mapping SLICE_X0 to CLBLL_L or CLBLM_M and SLICE_X1 to CLBLL_LL or CLBLM_L but it gets a bit confusing because the "L wires" map to the SLICE_X0 site inside a CLBLL and to the SLICE_X1 site inside the CLBLM.

If this is really such a big problem I think it would be better to dump the information into a separate JSON file which tile wire is connected to which site pin, instead of renaming stuff in a way so that it doesn't match the Xilinx names anymore.

It would be relatively easy to generate a JSON file like this one:

```
{
  "CLBLM_R": {
    "CLBLM_M_DI": [["SLICEM_X0", "DI"]],
    ....
  }
}
```

(`CLBLM_R` = tile name, `CLBLM_M_DI` = wire name, `SLICEM_X0` = normalized site name, `DI` = site pin)

I'm not claiming the the normalization for site names I use right now is perfect. For example, there is certainly an argument to be had to drop the M/L type suffix. But I don't think we should change it to something that is completely different from what Xilinx is using.

So coming full circle regarding the *what* and *why*: Afaict the *what* would be a change away from consistency with Xilinx names, and the *why* would be so we could avoid creating that easy-to-generate JSON file. I personally don't think that this *why* justifies the *what*, especially considering how easy it would be to simply generate the missing table.

----

> An alternative would be to change the names in the pseudo pip files as follows:

No! Please no! That would completely break consistency with the Xilinx wire names and make everything more complicated by orders of magnitudes without any clear advantage.

If we have to rename something and break consistency with Xilinx names, then please change the site name, not the wire names!

mithro, 2018-02-06:
Just checking - there is nothing called `SLICEM_X0` in Xilinx side of things right now? (Or are these names shown somewhere else?) There _are_ things named `SLICE_XnnYmm` where nn and mm refer to some random grid which seem to bare no relationship to reality?

Also confirming;
 * The `SLICE_XnnYnn` name are used for both `SLICEM` and `SLICEL` slices, so there is no way telling you what the type of the slice is just from the name?
 * Lastly the `SLICE_XnnYnn` name is used for both slices inside a CLB tile, so there is also no easy way to tell which slice it is inside the tile just from the name?

cliffordwolf, 2018-02-06:
Yes, the sites are called `SLICE_XnnYmm`. I don't quite understand what you mean by "bare no relationship to reality".

Like all sites, tiles, and and many of the other objects, the M and L slices build a common grid with the origin in the lower left corner. The two slices in a tile have the same Y coordinate and consecutive X coordinates. The one with the even X coordinate is the one we call X0, and the one with the odd X coordinate is the one we call X1 in our database. As I've said, there is an argument to be had to get rid of the M/L type suffix in our names.

There is nothing called e.g. SLICEM_X0 in the Xilinx database because sites are numbered independently from tiles. The object in the Xilinx database is called e.g. `SLICE_X9Y52` for tile `CLBLM_R_X7Y52`. But another CLBLM_R tile has completely different names for the sites. But for our tags we need to normalize all those names. So for example the tile name `CLBLM_R_X7Y52` becomes `CLBLM_R` in the database, and `SLICE_X9Y52` becomes `SLICEM_X0`, so that we have the same name for the same tag in two `CLBLM_R` tiles.

But (with the exception of the added type suffix for slices), we try to match the Xilinx names as closely as we can with our normalized names. For example, `CLBLM_R_X7Y52` is normalized to `CLBLM_R`, and not `RIGHT_MEMORY_CLB`. I think `SLICE_X9Y52` -> `SLICEM_X0` is a quite reasonable normalization, whereas with `SLICE_X9Y52` -> `CLBLM_M` it is not so obvious what is going on.

> The SLICE_XnnYnn name are used for both SLICEM and SLICEL slices, so there is no way telling you what the type of the slice is just from the name?

Yes. That is the reason why I added the M/L type suffix to our current normalized names.

> Lastly the SLICE_XnnYnn name is used for both slices inside a CLB tile, so there is also no easy way to tell which slice it is inside the tile just from the name?

I genuinely can't parse this sentence.

Yes, the SLICE_XnnYnn name is used for both slices inside a CLB tile. But I have no idea what you are asking in the 2nd part of the sentence.

mithro, 2018-02-06:
Thanks for clearing that all up!

Is there a way to tell if a slice is the first or second slice inside a CLB tile from the SLICE_XnnYnn name? Can you just `mod 2` the Y coordinate or something?

cliffordwolf, 2018-02-07:
You take the X coordinate mod 2, not the Y coordinate. Thus the suffix _X0 and _X1.

That's what I meant when I said that slices with even X coordinates are using the _X0 suffix and slices with odd X coordinates are using the _X1 suffix.

mithro, 2018-02-07:
Well... I'm an idiot. I totally didn't put together that despite Xilinx displaying the slices like this;
```
 Y1
 Y0
```
the slices are actually "side-by-side" like this;
```
 X0 X1
```

Then the choices of names `SLICE_X0` and `SLICE_X1` make a lot more sense because of the "X coordinate mod 2".

Let me think about this more. I think I'm now leaning towards your suggestion of of an extra JSON file you describe here;

> If this is really such a big problem I think it would be better to dump the information into a separate JSON file which tile wire is connected to which site pin, instead of renaming stuff in a way [...]

Do we currently have anything which describes any of the pins (IE tile / site / bel) at all?

kc8apf, 2018-02-07:
FWIW, Vivado uses additional properties of a site to distinguish things
like SLICEL vs SLICEM (see SITE_TYPE):

mithro, 2018-02-13:
Closing this in favor of #87.

closed 2018-02-13 by mithro