f4pga/prjxray #79: Create a virtualenv suitable for running Python tools

mithro, 2018-02-06:
It would be good to have a target in the makefile which set up a Python virtualenv (or what ever is the current state of the art for Python 3) and installed all the dependencies needed into it.

It would make sure that everyone has the same version of tools.

This gets more important as things like the documentation generation needs more Python modules which are not installed by default.

mithro, 2018-02-06:
Looks like this is already an issue, the version of yapf in Debian testing doesn't seem to support the options that are currently being set;
```
yapf: Unknown style option "SPLIT_BEFORE_EXPRESSION_AFTER_OPENING_PAREN"
yapf: Unknown style option "SPLIT_BEFORE_EXPRESSION_AFTER_OPENING_PAREN"
yapf: Unknown style option "SPLIT_BEFORE_EXPRESSION_AFTER_OPENING_PAREN"
```

galv, 2018-02-11:
I maintain a couple of python projects that use other languages as well. Some comments. 

Two options:

- conda
- virtualenv

conda provides virtual environments while also being a package manager.

Things to consider:

- Would you ever want to package this project via setup.py and make it pip-installable?
- Will you ever (like, *ever*) have to depend on a native dependency, compile a shared object against it, and load it in python? setuptools is awful at that, so you should use conda in that case. I believe pytorch uses conda for example. Conda would also presumably provide a path to fixing the gcc and cmake dependencies, if you felt that was important. (Note that I have not done that myself.)
- Conda installs its virtual environments at the user-level, not the project-level, by default. So if you ever had two prjxray git repos, you'd have to give the two different virtual environments distinct names, since they would get installed in, e.g., `~/miniconda2/envs`, rather than in a subdirectory of the repo.

Overall, virtualenv is probably enough along with a setup.py that lists requirements if you just want isolation.

mithro, 2018-02-11:
@galv Let's start with virtualenv and then move to conda if we need to later.

mithro, 2018-12-07:
This was done in https://github.com/SymbiFlow/prjxray/commit/4a94337c7931b02f863e761d4fc043bc02f7ff2c

closed 2018-12-07 by mithro