f4pga/prjxray #80: Should `make format` be running `yapf3`?

mithro, 2018-02-06:
Currently the make file does;
```
find . -name \*.py -and -not -path './third_party/*' -and -not -path './.git/*' -exec yapf -p -i {} \;
```

yapf seems to target Python 2 but we use Python3 everywhere?

```
$ apt-cache search yapf                                                                                                                                                                     
yapf - Python code formatter for different styles (Python 2)
yapf3 - Python code formatter for different styles (Python 3)
```

kc8apf, 2018-02-06:
I don't believe the version of Python used to execute YAPF makes any
difference.  I generally assume that Python dependencies are install via
pip and our requirements.txt file.

kc8apf, 2018-02-06:
https://github.com/google/yapf#python-versions says YAPF does change behavior depending on Python version it is execute under.  In that case, we should ensure that python3 is used by everything in our repo.  #79 seems like the right way to fix that.  Closing this as I don't anticipate needing to change the yapf invocation once virtualenv is in place.

closed 2018-02-06 by kc8apf