f4pga/prjxray #83: Generate `.json` output from the `.yaml` files

mithro, 2018-02-07:
Currently to understand the prjxray database you need to be able to parse both `.json` and `.yaml` file formats.

yaml has a canonical format way to be transformed into json -- so we should provide "jsonfied" version of the yaml files side-by-side with the yaml files.

This also makes it easier to use the data directly from javascript.

mcmasterg, 2018-02-07:
If this is something we care about, wouldn't we be better off standardizing on json? What is the main advantage of yaml over json?

mithro, 2018-02-07:
That is what I'm suggesting, that we have everything in json format.

mcmasterg, 2018-02-07:
You are proposing side by side. I'm proposing eliminating YAML

kc8apf, 2018-02-07:
I wrote the part files in YAML because I was planning to use the node aliasing feature.  It also happens that I found a decent YAML library for c++ before I found a decent JSON library.  Given other discussions about removing segments as a concept, needing to describe logical blocks with an N:M mapping onto tiles, etc, I'd like us to settle on a format that resolves those issues.

mithro, 2018-12-31:
This now happens.

closed 2018-12-31 by mithro