f4pga/prjxray #85: Document the timing information needed for timing driven place and route.

mithro, 2018-02-09:
Verilog-To-Routing supports timing driven place and route. For this we need timing information of all the parts and interconnects.

mithro, 2018-02-09:
@mcmasterg has started work on this.

mithro, 2019-02-08:
@JohnDMcMaster - Did you ever write this down somewhere?

litghost, 2019-04-17:
The routing model required by VPR is a RC tree with allocation of additional delay at the switches.  VPR computes the final delay using [Elmore delay](https://en.wikipedia.org/wiki/Elmore_delay).  It is unclear what the underlying model Vivado uses, but it is likely something similar.  

As a first step, I'm going to isolate a handful of timing paths and attempt to reproduce Vivado's ``get_timing_arc`` output using a fitted RC tree model.

litghost, 2019-04-17:
Quick test FF -> LUT:

Fanout = 1
```
report_property [get_net_delays -interconnect_only -to [get_pins lut/I0] -of_objects [get_nets FF_Q]]
Property   Type    Read-only  Value
CLASS      string  true       net_delay
ESTIMATED  bool    true       0
FAST_MAX   int     true       369
FAST_MIN   int     true       308
NAME       string  true       FF_Q_to_lut/I0
NET        string  true       FF_Q
SLOW_MAX   int     true       842
SLOW_MIN   int     true       694
TO_PIN     string  true       lut/I0
```

Fanout 2:
```
report_property [get_net_delays -interconnect_only -to [get_pins lut/I0] -of_objects [get_nets FF_Q]]
Property   Type    Read-only  Value
CLASS      string  true       net_delay
ESTIMATED  bool    true       0
FAST_MAX   int     true       349
FAST_MIN   int     true       288
NAME       string  true       FF_Q_to_lut/I0
NET        string  true       FF_Q
SLOW_MAX   int     true       822
SLOW_MIN   int     true       674
TO_PIN     string  true       lut/I0
```

Fanout 6:
```
report_property [get_net_delays -interconnect_only -to [get_pins lut/I0] -of_objects [get_nets FF_Q]]
Property   Type    Read-only  Value
CLASS      string  true       net_delay
ESTIMATED  bool    true       0
FAST_MAX   int     true       369
FAST_MIN   int     true       308
NAME       string  true       FF_Q_to_lut/I0
NET        string  true       FF_Q
SLOW_MAX   int     true       842
SLOW_MIN   int     true       694
TO_PIN     string  true       lut/I0
```

litghost, 2019-04-17:
So in this simple example, FAST_MIN varies by 10%

mithro, 2019-04-17:
Does it match a formula like `Tdel0 + N * Tdeln`?

litghost, 2019-04-17:
Observations from data above:
 - SLOW_MAX - SLOW_MIN and FAST_MAX - FAST_MIN is constant for fanout.  This implies that the capacitance difference in fanout is fixed.  I believe this makes physical sense, as the difference between SLOW_MAX and SLOW_MIN is accounting for differences in the Tdel of the switch, and not R or C differences in the wires.  The FAST vs SLOW *does* have a difference for fanout, implying that wire characteristics are changing between FAST and SLOW parts.

elmsfu, 2019-04-17:
If I understand there are three dimensions: process corner (fast/slow), production variation(max/min), and fanout. Delay is of the form `Tdel + R*C` where Tdel depends on the process corner and production variation, R is independent, and C depends on fanout.

I thought that R and C may depend on process corner, but it looks like the total delays are separable into a portion that depends on process corner and production variation and second component that depends on fanout only. If I understand correctly, only C should depend on fanout.

I guess the next question is if the fanout delay can be expressed in R*C with only C varying regularly as fanout increases. Looking at these data points I don't see an obvious solution. Going from fanout 1 to 2, we see a delay increase of 9, from 1 to 6 increase of 29.

Maybe I have a naive understanding of how C should change as fanout increases?

closed 2019-10-17 by litghost