f4pga/prjxray #87: Create a fuzzer for site and bel pins, plus internal psuedo pips and connections

mithro, 2018-02-12:
As suggested by @cliffordwolf in #74 it would be great to have the following details in the database;

 * Tile pins
 * Site pins
 * BEL pins
 * Routing muxes
 * Permanent wires between pins as psuedo "always" pips.

This would remove any need for us to match things via names in the SymbiFlow-arch-defs and allow a lot more things to be auto generated.

mcmasterg, 2018-02-13:
Sounds like you mean site wires

mithro, 2018-02-13:
Yeah, I was suggesting that we call "site wires" - "psuedo always pips" but you have convinced me that is a bad idea.

So basically I'm after;

 * File with all the pin information (Site, Tile, BEL, Routing BEL)
 * File with the site wire information (IE Site->Tile, Tile->BEL, BEL->Routing BEL, etc).

Does that make sense?

mcmasterg, 2018-10-18:
obsolete

closed 2018-10-18 by mcmasterg