f4pga/prjxray #120: Formatting tool versions (ex: yapf) drifting

mcmasterg, 2018-09-18:
I'm having issues getting consistent formatting for check in due to this drifting. From what I can tell, I originally had yapf 0.20.1 installed, and then tried upgrading to latest (0.24.0) since that's what the build server seems to do. However, this made things worse.

```
$ cat requirements.txt 
futures
sphinx
sphinx_rtd_theme
sphinx-autobuild
yapf
```

Should these versions, yapf in particular, be fixed?

Similarly, looks like issues with clang-format.

mithro, 2018-09-18:
How did it make things worse?

```
Collecting yapf (from -r requirements.txt (line 2))
/usr/local/lib/python2.7/dist-packages/pip/_vendor/requests/packages/urllib3/util/ssl_.py:122: InsecurePlatformWarning: A true SSLContext object is not available. This prevents urllib3 from configuring SSL appropriately and may cause certain SSL connections to fail. You can upgrade to a newer version of Python to solve this. For more information, see https://urllib3.readthedocs.io/en/latest/security.html#insecureplatformwarning.
  InsecurePlatformWarning
  Downloading https://files.pythonhosted.org/packages/32/12/9f6e437f6c3a76ae18f625998dcff92a51deef0294e1ef6493fbd144b08a/yapf-0.24.0-py2.py3-none-any.whl (168kB)
    97% |███████████████████████████████▏| 163kB 17.9MB/s eta 0:00:01    100% |████████████████████████████████| 174kB 5.9MB/s
Installing collected packages: futures, yapf
Successfully installed futures-3.2.0 yapf-0.24.0
```

mcmasterg, 2018-09-18:
I am on:

```
* master                                              8a257d4 [origin/master] Merge pull request #117 from mcmasterg/tiledoc

$ yapf --version
yapf 0.24.0
```

After running make format:
```
$ git reset --hard HEAD
$ make format
find . -name \*.cc -and -not -path './third_party/*' -and -not -path './.git/*' -exec clang-format -style=file -i {} \;
find . -name \*.h -and -not -path './third_party/*' -and -not -path './.git/*' -exec clang-format -style=file -i {} \;
find . -name \*.py -and -not -path './third_party/*' -and -not -path './.git/*' -exec yapf -p -i {} \;
$ git diff
diff --git a/experiments/example/segdata.py b/experiments/example/segdata.py
index 0eb1bee..ea776dc 100644
--- a/experiments/example/segdata.py
+++ b/experiments/example/segdata.py
@@ -54,8 +54,8 @@ with open("carrydata.txt", "r") as f:
         assert line[0] not in carry
         carry[line[0]] = dict()
         for i, n in enumerate(
-                "CYINIT:ZRO:ONE:AX:CIN DI0:AX:O5 DI1:AX:O5 DI2:AX:O5 DI3:AX:O5".
-                split()):
+                "CYINIT:ZRO:ONE:AX:CIN DI0:AX:O5 DI1:AX:O5 DI2:AX:O5 DI3:AX:O5"
+                .split()):
             n = n.split(":")
             for k in n[1:]:
                 carry[line[0]]["CARRY_%s_MUX_%s" %
diff --git a/htmlgen/htmlgen.py b/htmlgen/htmlgen.py
index da0ed6a..93063a7 100755
--- a/htmlgen/htmlgen.py
+++ b/htmlgen/htmlgen.py
@@ -848,9 +848,11 @@ for tiletype in sorted(cfgbits.keys()):
                 line = " --><td>%s</td>" % (pip)
                 for bit in grp_bits:
                     c = "-"
-                    if bit in routebits[tiletype] and pip in routebits[tiletype][bit]:
+                    if bit in routebits[tiletype] and pip in routebits[
+                            tiletype][bit]:
                         c = "1"
-                    if bit in routezbits[tiletype] and pip in routezbits[tiletype][bit]:
+                    if bit in routezbits[tiletype] and pip in routezbits[
+                            tiletype][bit]:
                         c = "0"
                     line = "%s%s<td align=\"center\">%s</td>" % (c, line, c)
                 lines.append(line)
diff --git a/tools/bittool.cc b/tools/bittool.cc
index a43e807..770bb95 100644
--- a/tools/bittool.cc
+++ b/tools/bittool.cc
@@ -11,7 +11,7 @@ namespace xc7series = prjxray::xilinx::xc7series;
 
 struct Action {
 	std::string name;
-	std::function<int(int, char* [])> handler;
+	std::function<int(int, char*[])> handler;
 };
 
 int ListConfigPackets(int argc, char* argv[]) {
diff --git a/tools/simpleroute.py b/tools/simpleroute.py
index 8384421..dc1c288 100755
--- a/tools/simpleroute.py
+++ b/tools/simpleroute.py
@@ -84,8 +84,8 @@ def db_gen():
                         node_node_pip[src_node] = dict()
                     if dst_node not in reverse_node_node:
                         reverse_node_node[dst_node] = set()
-                    node_node_pip[src_node][
-                        dst_node] = "%s.%s.%s" % (tile, dst, src)
+                    node_node_pip[src_node][dst_node] = "%s.%s.%s" % (
+                        tile, dst, src)
                     reverse_node_node[dst_node].add(src_node)
 
     return type_to_tiles, grid_to_tile, nodes, node_node_pip, reverse_node_node
```

mcmasterg, 2018-09-18:
Hmm maybe this is a python 2 vs 3 issue?

JohnDMcMaster, 2018-11-16:
Sounds like we should use clang-format-5 instead of clang-format