f4pga/prjxray #127: tool build failure: YAML::RegEx::m_z may be used uninitialized in this function

mcmasterg, 2018-09-27:
```
$ make
[  1%] Building CXX object third_party/yaml-cpp/CMakeFiles/yaml-cpp.dir/src/regex_yaml.cpp.o
In file included from /home/mcmaster/prjxray/third_party/yaml-cpp/src/regex_yaml.cpp:1:0:
/home/mcmaster/prjxray/third_party/yaml-cpp/src/regex_yaml.h: In constructor ‘YAML::RegEx::RegEx(const string&, YAML::REGEX_OP)’:
/home/mcmaster/prjxray/third_party/yaml-cpp/src/regex_yaml.h:31:20: error: ‘<anonymous>.YAML::RegEx::m_z’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
 class YAML_CPP_API RegEx {
                    ^~~~~
cc1plus: all warnings being treated as errors
third_party/yaml-cpp/CMakeFiles/yaml-cpp.dir/build.make:518: recipe for target 'third_party/yaml-cpp/CMakeFiles/yaml-cpp.dir/src/regex_yaml.cpp.o' failed
make[2]: *** [third_party/yaml-cpp/CMakeFiles/yaml-cpp.dir/src/regex_yaml.cpp.o] Error 1
CMakeFiles/Makefile2:1873: recipe for target 'third_party/yaml-cpp/CMakeFiles/yaml-cpp.dir/all' failed
make[1]: *** [third_party/yaml-cpp/CMakeFiles/yaml-cpp.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```

mcmasterg, 2018-09-27:
working around this for now

```
$ git diff
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 13645ab..6e3d067 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -10,7 +10,7 @@ if(NOT CMAKE_BUILD_TYPE)
 endif()
 
 set(CMAKE_CXX_STANDARD 14)
-add_compile_options(-Wall -Werror)
+add_compile_options(-Werror)
 
 # Hack for missing option in cctz
 option(BUILD_TESTING "" OFF)
```

mithro, 2018-09-27:
@mcmasterg Can you test #131 fixes the issue for you?

PR #131 (2018-10-02): cmake: Only add `-Wall and -Werror` for our code.
Fixes #127.
commit 8e6933fe (2018-10-02): cmake: Only add `-Wall and -Werror` for our code.

Fixes #127.
closed 2018-10-02 by mcmasterg