f4pga/prjxray #128: Create small partial reconfiguration ROI for VPR testing

mcmasterg, 2018-09-27:
Current ROI is 50 by 8 CLBs (400 CLBs). We'd like something more in the ballpark of 10x10 for early testing

mcmasterg, 2018-09-27:
[image]

@litghost : proposing 12 CLB rows by 8 CLB columns ROI => 96 CLB tiles. 2 inputs, 2 outputs, and clock. Is this good enough for a simple test? This can be done with pretty low work on my side. I might be able to increase the I/O with a small tweak if it would be helpful.

metadata from the attached image:

```
name node pin wire
clk CLK_HROW_TOP_R_X60Y130/CLK_HROW_CK_BUFHCLK_L0 F21 HCLK_VBRK_X34Y130/HCLK_VBRK_CK_BUFHCLK0
din[0] INT_R_X9Y100/NE2BEG3 G22 VBRK_X29Y106/VBRK_NE2A3
din[1] INT_R_X9Y103/NE2BEG3 G21 VBRK_X29Y109/VBRK_NE2A3
dout[0] INT_L_X10Y105/WW2BEG0 D21 VBRK_X29Y110/VBRK_WW2A0
dout[1] INT_L_X10Y108/WW2BEG0 E21 VBRK_X29Y113/VBRK_WW2A0
```

litghost, 2018-09-27:
I think having more outputs would be useful.  We should also make sure that the outputs go to LED's on the test platform.  I believe we agreed on a arty?

mcmasterg, 2018-09-27:
More => how many?

litghost, 2018-09-27:
5-8 would be fine.  Given how the INT_R/INT_L work, I don't imagine it would be too bad.

litghost, 2018-09-27:
> If I make it 8 inputs and 8 outputs, I think I could make it 17x6 (102 CLBs) without too much effort. If that works, would that suffice?

Ya, I think that sounds good.

mcmasterg, 2018-10-01:
[image]

How about this for an ROI (6x by 18y CLBs => 108)? 8 outputs at top, then 8 inputs at bottom. Clock meanders in at top

```
        # 4 switches then 4 buttons
        set sw_but "A8 C11 C10 A10  D9 C9 B9 B8"
        # 4 LEDs then 4 RGB LEDs (green only)
        set leds "H5 J5 T9 T10  F6 J4 J2 H6"

        # 100 MHz CLK onboard
        set pin "E3"
```

litghost, 2018-10-01:
That's perfect

closed 2018-10-05 by litghost