f4pga/prjxray #134: The way sites are named in Tile.get_instance_sites is broken

litghost, 2018-10-02:
The way that Tile.get_instance_sites in the prjxray library gets the names for instance sites is broken on the following site types:

- BRAM_L
- BRAM_R
- HCLK_IOI3
- CMT_TOP_L_UPPER_B
- CMT_TOP_R_UPPER_B

Currently Tile.get_instance_sites isn't used, and none of the sites in those site types are currently being integrated, so this issue doesn't need to be solved immediately.  However for debugging purposes, it would be good to be able to go from the "generic" database files to the specific names that the database was originally generated from.

litghost, 2019-02-01:
This actually turns out to be important because of how symbiflow-arch-defs imports the BRAM tile.  This issue must be fixed for stable naming, so symbiflow-arch-defs can reliably name BRAM signals.

litghost, 2019-02-01:
Example: https://github.com/SymbiFlow/symbiflow-arch-defs/blob/master/xc7/primitives/bram_l/bram_l.pb_type.xml#L106

RAMB18_X0Y29_WREN is based on the site base name, which is currently unstable as a result of the underlying problem.

litghost, 2019-02-01:
Example of the instability: https://github.com/SymbiFlow/prjxray-db/commit/b2c6e1c6c1693204fbaa1ff974670d4401f66579#diff-441f680d8b2660ab9a48a8479cd9743dL7189

tmichalak, 2019-02-12:
Some useful notes for further efforts:
tile_type_BRAM_L.json (https://github.com/SymbiFlow/prjxray-db/commit/b2c6e1c6c1693204fbaa1ff974670d4401f66579#diff-441f680d8b2660ab9a48a8479cd9743dL7189)
gets generated by https://github.com/SymbiFlow/prjxray/blob/master/fuzzers/074-dump_all/reduce_tile_types.py
The logic in https://github.com/SymbiFlow/prjxray/blob/master/fuzzers/074-dump_all/reduce_tile_types.py#L162 is the problem
https://github.com/SymbiFlow/prjxray/blob/master/fuzzers/074-dump_all/reduce_tile_types.py#L176 is wrong
The reduction method doesn't work as the idea is that no matter what tile you choose, get_sites should return the same result. So the particular output is not important it just needs to be stable, and possible to go from instance to generic and back.

litghost, 2019-02-13:
To close this issue, need to add a test that ensures that Tile.get_instance_sites works correctly (e.g. generic to specific site names).

acomodi, 2019-02-14:
I am testing it and the get_instance_sites has to be slightly modified. I'll shortly make a PR with the changes.

closed 2019-02-14 by litghost