f4pga/prjxray #136: Timing fuzzer: min corners tend towards 0 delay

mcmasterg, 2018-10-03:
When solving for the minimum delay, an ideal min delay is equal to the max delay. However, its legal to move a delay as low as 0. The timing very disproportionately assigns minimum delays to 0 vs what is expected to be realistic. For example:

```
Corner fast_max
  pips: 2882 / 8482 solved, 3978 / 8482 covered
Corner fast_min
  pips: 40 / 8482 solved, 3978 / 8482 covered
```

Investigate what is causing this pressure, and if it can be fixed.

mithro, 2019-06-12:
@kgugala / @litghost - I think this is no longer relevant with the new way we get timing data?

kgugala, 2019-06-12:
@mithro I think this issue is related to wires delays. We still use the same method for calculating the wires delays, so it may be still relevant

litghost, 2019-06-12:
> @kgugala / @litghost - I think this is no longer relevant with the new way we get timing data?

Ya, this is no longer relevant.

closed 2019-07-30 by litghost