f4pga/prjxray #137: Timing fuzzer: excessive disk space usage

mcmasterg, 2018-10-04:
The fuzzer spits out a pretty exhaustive netlist of the current design to timing4.txt. When the site wires were added in, this added a significant amount of generally boring routing. Find a way to reduce these and/or look at other wastes

Known issues / suggestions:
* nodes are no longer needed. Eliminate them from the file
* Many site wires are pretty boring and redundant. Consider adding a hash map with something like (delays, site_type, bel_type, bel_pin) and only add if its new

Here are some sizes from oneblinkw for an example:
* timing4i.csv: 372K
* timing4s.csv: 12K
* timing4.txt: 132M

That is, the vivado output is quickly chopped down from 132 MB to about 0.4 MB, a significant reduction. Additionally, we really want to operate on a much bigger design with many input specimens, so the disk space will grow infeasible very quickly.

Maybe there is a quick hack we could do to pipe it directly from vivado into python?

mcmasterg, 2018-10-04:
A quick tests shows that removing bel wires has negligible impact on the output size. Think the real reason why the disk space increased was that I wasn't handling a bunch of cases correctly before, but now those nets are being output

mithro, 2019-06-12:
@kgugala / @litghost - I think this is no longer relevant with the new way we get timing data?

kgugala, 2019-06-12:
I got hit by this quite recently. I'll check if it still happens.

@mithro FYI in CI we run simpler version, so we do not get hit by out of space error

litghost, 2019-06-12:
The original 007 fuzzer is no longer used, and can be removed.  We should keep @kgugala BEL timing fuzzer, and the routing fuzzer is in 074.

closed 2019-07-30 by litghost