f4pga/prjxray #140: tilegrid.json: Update version utilities to use latest format

litghost, 2018-10-05:
The tilegrid.json conversion was incomplete, the following utilities relied on the old format:

- fasm2frame.py
- segprint2fasm.py
- segprint.py

#139 adds bridge fields and code to solve this in the short term, but in the long term utilities should likely converge.  Suggestion to resolve is to add cabilities to the prjxray library that hide the underlying file format from the various utilities, and expose a usable API for interacting with the underlying database.

mcmasterg, 2018-10-18:
https://github.com/SymbiFlow/prjxray/pull/159 makes some progress on this

litghost, 2018-10-22:
#167 converts fasm2frame.py and bits2fasm.py

mcmasterg, 2018-10-22:
Close this? There's a little lingering in htmlgen, but that has a dedicated issue (https://github.com/SymbiFlow/prjxray/issues/173)

litghost, 2018-10-23:
utils/segprint.py is not converted yet.

closed 2019-01-08 by litghost