f4pga/prjxray #146: Fuzzers should diff entire bitstream

mcmasterg, 2018-10-10:
For size and/or simplicity we originally were filtering out to only dump a smallish area of the bitstream. However, this is causing issues for things like the BRAM fuzzer. Its also not saving us that much disk space anymore. For example, my test bitstream increased from 208 KB to 212 KB, basically negligible.

However, we may add additional aliased bits if we add this. So we'll have to re-run the fuzzers and verify everything is still good.

```
mcmaster@xc2064:~/prjxray/fuzzers/005-tilegrid/specimen_001$ /home/mcmaster/prjxray/build/tools/bitread --part_file /home/mcmaster/prjxray/database/kintex7/xc7k70tfbg676-2.yaml -F 0x00400100:0x004006ff -o design_SLICE_X9Y50.bits -z -y design_RAMB36_X0Y10.bit
Bitstream size: 3157171 bytes
Config size: 789255 words
Number of configuration frames: 7432
DONE
mcmaster@xc2064:~/prjxray/fuzzers/005-tilegrid/specimen_001$ /home/mcmaster/prjxray/build/tools/bitread --part_file /home/mcmaster/prjxray/database/kintex7/xc7k70tfbg676-2.yaml -F 0x00000000:0xffffffff -o design_SLICE_X9Y50_full.bits -z -y design_RAMB36_X0Y10.bit
Bitstream size: 3157171 bytes
Config size: 789255 words
Number of configuration frames: 7432
DONE
mcmaster@xc2064:~/prjxray/fuzzers/005-tilegrid/tmp$ du -hs design_SLICE_X9Y50*.bits
208K    design_SLICE_X9Y50.bits
212K    design_SLICE_X9Y50_full.bits
```

cliffordwolf, 2018-10-11:
> However, we may add additional aliased bits if we add this.

jfyi: This is the main reason why the current code is only looking at certain sections of the bit stream.

mcmasterg, 2018-10-18:
A7 is diffing entire bitstream now. We have some unrelated fuzzer regressions I'm working through, will look into this more as part of that

closed 2018-11-06 by mithro

mcmasterg, 2018-11-06:
Filed https://github.com/SymbiFlow/prjxray/issues/230 Think issue is not related to this one