f4pga/prjxray #151: prjxray/htmlgen/htmlgen.py fails with KeyError: 'segments'

mithro, 2018-10-14:
https://travis-ci.org/SymbiFlow/prjxray-db/builds/441397423
```
source $(dirname ${BASH_SOURCE[0]})/../../utils/environment.sh
--------------------------------------------
Loading tilegrid.
Writing ./html/artix7/index.html
Traceback (most recent call last):
  File "/tmp/tmp.0GUYED3cEh/prjxray/htmlgen/htmlgen.py", line 322, in <module>
    segdata = grid["segments"][tiledata["segment"]]
KeyError: 'segments'
```

JohnDMcMaster, 2018-10-14:
This was my comment the other day that there are a number of tools that use the tilegrid database are broken. They will be cleaned up as part of my changes that are also breaking the database

mcmasterg, 2018-10-18:
fixed

closed 2018-10-18 by mcmasterg