f4pga/prjxray #152: fuzzers/015-clbnffmux fails with kintex7 build

mithro, 2018-10-14:
When trying to build the kintex7 database, the following error occurs;
```
Compiling segment data.
Traceback (most recent call last):
  File "../generate.py", line 78, in <module>
    segmk.compile(bitfilter=bitfilter)
  File "../../../utils/segmaker.py", line 226, in compile
    add_site_tags()
  File "../../../utils/segmaker.py", line 174, in add_site_tags
    segments, segname, tiledata, bitfilter=bitfilter)
  File "../../../utils/segmaker.py", line 151, in add_segbits
    bitname_bit):
  File "../generate.py", line 74, in bitfilter
    assert os.getenv("XRAY_DATABASE") == "artix7"
AssertionError
Makefile:16: recipe for target 'specimen_001/OK' failed
make[2]: *** [specimen_001/OK] Error 1
make[2]: Leaving directory '/big/github/SymbiFlow/prjxray/fuzzers/015-clbnffmux'
Makefile:20: recipe for target 'run' failed
make[1]: *** [run] Error 2
make[1]: Leaving directory '/big/github/SymbiFlow/prjxray/fuzzers/015-clbnffmux'
Makefile:16: recipe for target '015-clbnffmux/run.ok' failed
make: *** [015-clbnffmux/run.ok] Error 2
```

mithro, 2018-10-14:
After doing the following;
```patch
diff --git a/fuzzers/015-clbnffmux/generate.py b/fuzzers/015-clbnffmux/generate.py
index 79b4991..a91e9a7 100644
--- a/fuzzers/015-clbnffmux/generate.py
+++ b/fuzzers/015-clbnffmux/generate.py
@@ -71,7 +71,7 @@ for loc, muxes in cache.items():

 # we know that all bits for those MUXes are in frames 30 and 31, so filter all other bits
 def bitfilter(frame_idx, bit_idx):
-    assert os.getenv("XRAY_DATABASE") == "artix7"
+    assert os.getenv("XRAY_DATABASE") in ("artix7", "kintex7")
     return frame_idx in [30, 31]

diff --git a/fuzzers/016-clbnoutmux/generate.py b/fuzzers/016-clbnoutmux/generate.py
index ede555f..a33e536 100644
--- a/fuzzers/016-clbnoutmux/generate.py
+++ b/fuzzers/016-clbnoutmux/generate.py
@@ -70,7 +70,7 @@ for loc, muxes in cache.items():

 def bitfilter(frame_idx, bit_idx):
-    assert os.getenv("XRAY_DATABASE") == "artix7"
+    assert os.getenv("XRAY_DATABASE") in ("artix7", "kintex7")

     # locations of A5MA, B5MA, C5MA, D5MA bits. because of the way we generate specimens
     # in this fuzzer we get some aliasing with those bits, so we have to manually exclude
diff --git a/fuzzers/018-clbram/generate.py b/fuzzers/018-clbram/generate.py
index 88b99c8..387e950 100644
--- a/fuzzers/018-clbram/generate.py
+++ b/fuzzers/018-clbram/generate.py
@@ -132,7 +132,7 @@ for l in f:
 def bitfilter(frame_idx, bit_idx):
     # Hack to remove aliased PIP bits on CE
     # We should either mix up routing more or exclude previous DB entries
-    assert os.getenv("XRAY_DATABASE") == "artix7"
+    assert os.getenv("XRAY_DATABASE") in ("artix7", "kintex7")
     return (frame_idx, bit_idx) not in [(0, 27), (1, 25), (1, 26), (1, 29)]

```

```
make[2]: Entering directory '/big/github/SymbiFlow/prjxray/fuzzers/018-clbram'
bash generate.sh specimen_001
+ source /big/github/SymbiFlow/prjxray/utils/genheader.sh
++ '[' -z kintex7 ']'
++ set -ex
++ test 1 = 1
++ test '!' -e specimen_001
++ mkdir specimen_001
++ cd specimen_001
+ python3 ../top.py
Traceback (most recent call last):
  File "../top.py", line 108, in <module>
    loc = next(slices)
  File "../top.py", line 58, in gen_slicems
    assert ((12, 28) == SLICEX)
AssertionError
Makefile:15: recipe for target 'specimen_001/OK' failed
make[2]: *** [specimen_001/OK] Error 1
make[2]: Leaving directory '/big/github/SymbiFlow/prjxray/fuzzers/018-clbram'
Makefile:19: recipe for target 'run' failed
make[1]: *** [run] Error 2
make[1]: Leaving directory '/big/github/SymbiFlow/prjxray/fuzzers/018-clbram'
Makefile:19: recipe for target '018-clbram/run.ok' failed
make: *** [018-clbram/run.ok] Error 2
```

mcmasterg, 2018-10-18:
I think the XRAY_DATABASE assertions can be removed: we have reasonable confidence at this point that bitstream definitions are portable across 7 series