f4pga/prjxray #154: Sort the json output files

mithro, 2018-10-16:
Currently the output of files like [`tile_type_BRAM_L.json`](https://github.com/SymbiFlow/prjxray-db/blob/master/artix7/tile_type_BRAM_L.json) are unsorted, see below;
```json
    {
      "type": "RAMB18E1",
      "name": "X0Y30",
      "y_coord": 30,
      "prefix": "RAMB18",
      "x_coord": 0,
      "site_pins": {
        "WRCOUNT1": "BRAM_RAMB18_WRCOUNT1",
        "WRCOUNT8": "BRAM_RAMB18_WRCOUNT8",
        "ADDRARDADDR2": "BRAM_RAMB18_ADDRARDADDR2",
        "RDCOUNT4": "BRAM_RAMB18_RDCOUNT4",
        "DIBDI8": "BRAM_RAMB18_DIBDI8",
        "ADDRATIEHIGH0": "BRAM_RAMB18_ADDRATIEHIGH0",
        "RDCOUNT9": "BRAM_RAMB18_RDCOUNT9",
```

It would be good if this output was sorted / canonicalized in some way.

mithro, 2018-10-16:
Normally you use something like;
```python
json.dump(j, fout, sort_keys=True, indent=4, separators=(',', ': '))
```

mithro, 2018-10-28:
The issue actually seems to mostly be the lists of wire pairs.

```
+        "wire_pairs": [
+            [
+                "CLBLL_EE2A0",
+                "VFRAME_EE2A0"
+            ],
+            [
+                "CLBLL_EE2A1",
+                "VFRAME_EE2A1"
+            ],
```

mithro, 2018-12-07:
I'm currently using the following Python to sort the output;
```python
#!/usr/bin/env python3
import json
import sys
data = json.load(sys.stdin)

def walker(o, f):
    if isinstance(o, dict):
        for i in o.values():
            walker(i, f)
    elif isinstance(o, list):
        for i in o:
            walker(i, f)
    f(o)

def f(o):
    if isinstance(o, list):
        if len(o) > 2:
            strings = all(isinstance(x, str) for x in o)
            if strings:
                print("Sorted:", repr(o)[:100], file=sys.stderr)
                o.sort()

walker(data, f)

json.dump(data, sys.stdout, sort_keys=True, indent=4)
```

mithro, 2018-12-07:
I'm now using;
```python
#!/usr/bin/env python3
import re
import json
import sys
data = json.load(sys.stdin)

if isinstance(data, list):
    def extract_numbers(s):
        """
        >>> extract_numbers("CLK_HROW_WR10END2_3")
        ('CLK_HROW_WR', 10, 'END', 2, '_', 3)
        >>> extract_numbers("VBRK_WR1END2")
        ('VBRK_WR', 1, 'END', 2)
        """
        bits = []
        for m in re.finditer("([^0-9]*)([0-9]*)",s):
            if m.group(1):
                bits.append(m.group(1))
            if m.group(2):
                bits.append(int(m.group(2)))
        return tuple(bits)

    print("  Tileconn!", file=sys.stderr)

    for o in data:
        o['wire_pairs'].sort(key=lambda o: (extract_numbers(o[0]), extract_numbers(o[1])))

    data.sort(key=lambda o: (o['tile_types'], o['grid_deltas']))
else:
    def walker(o, f):
        if isinstance(o, dict):
            for i in o.values():
                walker(i, f)
        elif isinstance(o, list):
            for i in o:
                walker(i, f)
        f(o)

    def f(o):
        if isinstance(o, list):
            if len(o) > 2:
                strings = all(isinstance(x, str) for x in o)
                if strings:
                    print("  Sorted:", repr(o)[:100], file=sys.stderr)
                    o.sort()

    walker(data, f)

json.dump(data, sys.stdout, sort_keys=True, indent=4)
```

mcmasterg, 2018-12-10:
Can you merge this in somehow? Ideally in 074 (IIRC thats what generates these)

commit db7e2d7d (2018-12-18): Adding tool for canonicalizing json files.

Fixes #154.
closed 2018-12-18 by mcmasterg