f4pga/prjxray #155: Make bits2fasm

JohnDMcMaster, 2018-10-17:
Right now we use segprint2fasm to create fasm from a bitstream.  segprint was intended as a debugging utility and not to be used this way. Instead we should just make a dedicated bits2fasm utility and eliminate the intermediate step. I've recently cleaned up segprint, so this should be a lot easier

litghost, 2018-10-17:
I've done some work on this in https://github.com/litghost/prjxray/tree/fasm_refactor

https://github.com/litghost/prjxray/blob/fasm_refactor/minitests/roi_harness/bit2fasm.sh

mcmasterg, 2018-10-17:
Talked to @litghost , there will be some conflicts but we'll review them after. sample output from the new tool: https://gist.github.com/mcmasterg/380d99efbb5bb5ac6fa6b5dbde167bcd

I copied over the formatting code, so I think it should be the same, but you never know

mcmasterg, 2018-10-18:
fixed

closed 2018-10-18 by mcmasterg