f4pga/prjxray #157: Latest Artix 7 ROI breaks the 018-clbram fuzzer

mithro, 2018-10-17:
```
make[2]: Entering directory '/big/github/SymbiFlow/prjxray/fuzzers/018-clbram'
bash generate.sh specimen_001
+ source /big/github/SymbiFlow/prjxray/utils/genheader.sh
++ '[' -z artix7 ']'
++ set -ex
++ test 1 = 1
++ test '!' -e specimen_001
++ mkdir specimen_001
++ cd specimen_001
+ python3 ../top.py
Traceback (most recent call last):
  File "../top.py", line 108, in <module>
    loc = next(slices)
  File "../top.py", line 58, in gen_slicems
    assert ((12, 28) == SLICEX)
AssertionError
Makefile:15: recipe for target 'specimen_001/OK' failed
make[2]: *** [specimen_001/OK] Error 1
make[2]: Leaving directory '/big/github/SymbiFlow/prjxray/fuzzers/018-clbram'
Makefile:19: recipe for target 'run' failed
make[1]: *** [run] Error 2
make[1]: Leaving directory '/big/github/SymbiFlow/prjxray/fuzzers/018-clbram'
Makefile:19: recipe for target '018-clbram/run.ok' failed
make: *** [018-clbram/run.ok] Error 2
```

mithro, 2018-10-17:
Maybe;
```patch
diff --git a/fuzzers/018-clbram/top.py b/fuzzers/018-clbram/top.py
index b2cac6c..2024b0f 100644
--- a/fuzzers/018-clbram/top.py
+++ b/fuzzers/018-clbram/top.py
@@ -55,8 +55,8 @@ def gen_slicems():
         x = 6, 8, 10, 12, 14
     '''
     # TODO: generate this from DB
-    assert ((12, 28) == SLICEX)
-    for slicex in (12, 14):
+    #assert ((12, 28) == SLICEX), repr(SLICEX)
+    for slicex in (8, 12, 14):
         for slicey in range(*SLICEY):
             # caller may reject position if needs more room
             #yield ("SLICE_X%dY%d" % (slicex, slicey), (slicex, slicey))
```

mithro, 2018-10-17:
The comment above says;
```python
'''
x
   Always even (ie 100, 102, 104, etc)
   In our ROI
   x = 6, 8, 10, 12, 14
'''
```

mcmasterg, 2018-10-18:
also note : ./019-ndi1mux/top.py:41: assert ((12, 28) == SLICEX)

mcmasterg, 2018-10-18:
019 fix: https://github.com/SymbiFlow/prjxray/pull/163

mcmasterg, 2018-10-18:
The core issues should be fixed. I'm going to also replace anywhere that has slice_xy(), although I think because most of these use regular expressions, they shouldn't be affected by the addition of the BRAM and DSP strings

closed 2018-10-18 by mcmasterg