f4pga/prjxray #169: Rethink ROI usage

mcmasterg, 2018-10-22:
We have three things right now that the ROI are used for:
- The area we target for VPR PnR testing
- The area that a developer wants to run for quick testing
- The area that we want the database to contain, which is now arguably the full chip area (this is why 074 exists)

We should rethink these in order to clean up the code. Proposal is something like:
- Expand "official" ROI to contain all tile addresses. However, keep the SLICEs and such limited to just the VPR area for now. This will make it so the database contains all tiles, but only addresses for the VPR ROI
- Have a secondary ROI for VPR testing. This is also probably a good area to use for quick test development

litghost, 2018-10-22:
I think this makes sense to me.  This work is directly tied with the change of the design.txt output from the roi_harness to support configurable grid roi's for VPR.  Once design.txt (e.g. design.json) also describes what tiles VPR should use, we can totally decouple these 3 things.

JohnDMcMaster, 2018-11-16:
this was fixed in a recent PR

closed 2018-11-16 by JohnDMcMaster