f4pga/prjxray #171: All output products should go into "build" dir

mcmasterg, 2018-10-22:
Right now we dump misc output products into source directories. It would be nice to instead consistently dump all products into a unified output directory

mcmasterg, 2018-11-28:
Following are left:
  * 051-055
  * 057, 058
  * 071-074
  * 100

mcmasterg, 2018-12-10:
Following are left:
  * 055
  * 057, 058
  * 071-074
  * 100

mcmasterg, 2018-12-20:
Following are left:
* 058
* 071-074

JohnDMcMaster, 2019-01-07:
Following are left: 071-074

mcmasterg, 2019-01-08:
nats looking at this, adding some details discussed over IRC:

```
I'd like there to only be two generated files in fuzzer directories: build, run.ok
(with the eventual goal to eliminate run.ok, but we can do that later)
This is the case on most fuzzers today, except for the three mentioned in the ticket
You will need to primarily edit the makefile, but might also need to adjust some paths in say tcl or python scripts in that directory
the idea is that if everything is moved to build, simply deleting the build dir will clean it
```

natsfr, 2019-01-09:
Following left 072-074, I have patches but because of RAM use problem in 072 I can't test them for 072 **AND** 074 (dependent on 072).
Discussing that at the moment to figure out what to do next.

natsfr, 2019-01-14:
Should be ok to close now

JohnDMcMaster, 2019-01-14:
Looks like part-yaml still needs this

JohnDMcMaster, 2019-01-14:
fixed

closed 2019-01-14 by JohnDMcMaster