f4pga/prjxray #181: Create tool to patch blockram contents in bitstream

mithro, 2018-10-23:
Allow easy swapping of ROM / RAM contents in a bistream using only open tooling?

mithro, 2018-10-23:
Probably need some way to produce annotations for were various RAM contents ends up?

mithro, 2018-10-23:
Could this be another example of FASM machine parsable comments that don't end up as actual bitstream values?

litghost, 2019-03-05:
FASM can now process BRAM data section bits correctly, so this is basically just creating a tool to convert data bits into FASM output. There needs to be section mapping of address lines to tile locations of some sort.

mithro, 2019-04-08:
Would be good if a tool could also patch DRAM contents too!

tmichalak, 2021-01-26:
BRAM patcher is being worked on in a separate [repo](https://github.com/SymbiFlow/prjxray-bram-patch)

closed 2021-01-26 by tmichalak