f4pga/prjxray #200: Fuzzers are broken with the Kintex 7 settings

mithro, 2018-10-27:
Changes to the fuzzers mean they no longer work with the [Kintex 7 settings](https://github.com/SymbiFlow/prjxray/blob/master/database/kintex7/settings.sh).

So far the following fuzzers work;

 - [ ] 001-part-yaml
 - [ ] 005-tilegrid
 - [ ] 007-timing
 - [ ] 010-lutinit
 - [ ] 011-ffconfig
 - [ ] 012-clbn5ffmux
 - [ ] 013-clbncy0
 - [ ] 014-ffsrcemux
 - [ ] 015-clbnffmux
 - [ ] 016-clbnoutmux
 - [ ] 017-clbprecyinit
 - [ ] 018-clbram
 - [ ] 019-ndi1mux
 - [ ] 050-intpips
 - [ ] 051-imuxlout
 - [ ] 052-clkin
 - [ ] 053-ctrlin
 - [ ] 054-gfan
 - [ ] 055-gnd
 - [ ] 056-rempips
 - [ ] 057-bipips
 - [ ] 058-hclkpips
 - [ ] 071-ppips
 - [ ] 072-ordered_wires
 - [ ] 073-get_counts
 - [ ] 074-dump_all
 - [ ] 100-dsp-mskpat
 - [ ] 102-bram-data

mithro, 2018-10-27:
The current failing fuzzer is [`005-tilegrid`](https://github.com/SymbiFlow/prjxray/tree/master/fuzzers/005-tilegrid) it fails with;
```
+ for x in design*.bit
+ /big/github/SymbiFlow/prjxray-kintex/build/tools/bitread --part_file /big/github/SymbiFlow/prjxray-kintex/database/kintex7/xc7k70tfbg676-2.yaml -F 0x00400100:0x004006ff -o design_SLICE_X6Y50.bits -z -y design_SLICE_X6Y50.bit
Bitstream size: 3157171 bytes
Config size: 789255 words
Number of configuration frames: 7432
DONE
+ for x in design*.bit
+ /big/github/SymbiFlow/prjxray-kintex/build/tools/bitread --part_file /big/github/SymbiFlow/prjxray-kintex/database/kintex7/xc7k70tfbg676-2.yaml -F 0x00400100:0x004006ff -o design_SLICE_X8Y50.bits -z -y design_SLICE_X8Y50.bit
Bitstream size: 3157171 bytes
Config size: 789255 words
Number of configuration frames: 7432
DONE
+ for x in design_*.bits
+ diff -u design.bits design_RAMB36_X0Y10.bits
+ grep '^[-+]bit'
Makefile:13: recipe for target 'specimen_001/OK' failed
make[2]: *** [specimen_001/OK] Error 1
make[2]: Leaving directory '/big/github/SymbiFlow/prjxray-kintex/fuzzers/005-tilegrid'
Makefile:17: recipe for target 'run' failed
make[1]: *** [run] Error 2
make[1]: Leaving directory '/big/github/SymbiFlow/prjxray-kintex/fuzzers/005-tilegrid'
Makefile:10: recipe for target '005-tilegrid/run.ok' failed
make: *** [005-tilegrid/run.ok] Error 2
```

mcmasterg, 2018-11-06:
that failure is due to the bitstream range. I'll create a PR