f4pga/prjxray #223: Kintex fails with 948 errors on 0740-dump_all fuzzer.

mithro, 2018-11-06:
The kintex database run is getting a lot closer now. It seems to be failing at the very end with the following error;
```
2018-10-29 03:09:14.308786 Writing node tree
+ python3 reduce_site_types.py --output_dir output
+ python3 generate_grid.py --root_dir specimen_001/ --output_dir output --ignored_wires kintex7_ignored_wires.txt
2018-10-29 03:10:47.391586 Running 10 processes
2018-10-29 03:10:47.579703 Reading tilegrid
2018-10-29 03:10:47.787103 Creating tile map
100% (24453 of 24453) |##########################################################################################################################################################################################################################################################################################################################################################################| Elapsed Time: 0:01:37 Time:  0:01:37
2018-10-29 03:12:52.867112 Reading node tree
2018-10-29 03:13:36.497577 Creating tile connections
100% (2663055 of 2663055) |######################################################################################################################################################################################################################################################################################################################################################################| Elapsed Time: 0:03:21 Time:  0:03:21
2018-10-29 03:17:11.411896 Writing tileconn
2018-10-29 03:17:12.594459 Connecting wires to verify tileconn
100% (8339126 of 8339126) |######################################################################################################################################################################################################################################################################################################################################################################| Elapsed Time: 0:05:25 Time:  0:05:25
2018-10-29 03:23:43.831873 Verifing tileconn
948 errors detected, see output/error_nodes.json for details.
Makefile:17: recipe for target 'specimen_001/OK' failed
make[2]: *** [specimen_001/OK] Error 1
make[2]: Leaving directory '/big/github/SymbiFlow/prjxray-kintex/fuzzers/074-dump_all'
Makefile:21: recipe for target 'run' failed
make[1]: *** [run] Error 2
make[1]: Leaving directory '/big/github/SymbiFlow/prjxray-kintex/fuzzers/074-dump_all'
```

mithro, 2018-11-06:
[`output/error_nodes.json` can be found here](https://drive.google.com/file/d/1qwi3WNw5IBVvhJTZXSTZa5t8ndyMhR7s/view?usp=sharing)

mithro, 2018-11-06:
analyze_errors returns;
```python
('LIOI3_TBYTESRC_X0Y119/LIOI_I2GCLK_TOP0', (('LIOI3_X0Y121/LIOI_I2GCLK_BOT1',), ('LIOI3_X0Y117/LIOI_I2GCLK_TOP1',)))
('RIOI_TBYTESRC_X43Y81/RIOI_I2GCLK_TOP0', (('RIOI_X43Y79/RIOI_I2GCLK_TOP1',), ('RIOI_X43Y83/RIOI_I2GCLK_BOT1',)))
('LIOI3_TBYTESRC_X0Y93/LIOI_I2GCLK_TOP0', (('LIOI3_X0Y91/LIOI_I2GCLK_TOP1',), ('LIOI3_X0Y95/LIOI_I2GCLK_BOT1',), ('CMT_TOP_R_UPPER_B_X8Y83/CMT_PHASER_UP_DQS_TO_PHASER_D',)))
('LIOI3_TBYTESRC_X0Y43/LIOI_I2GCLK_TOP0', (('LIOI3_X0Y45/LIOI_I2GCLK_BOT1',), ('CMT_TOP_R_UPPER_B_X8Y31/CMT_PHASER_UP_DQS_TO_PHASER_D',), ('LIOI3_X0Y41/LIOI_I2GCLK_TOP1',)))
('LIOI3_X0Y159/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTESRC_X0Y157/LIOI_I2GCLK_TOP1',),))
('RIOI_X43Y61/RIOI_I2GCLK_TOP0', (('RIOI_TBYTETERM_X43Y63/RIOI_I2GCLK_BOT1',),))
('RIOI_X43Y55/RIOI_I2GCLK_TOP0', (('RIOI_TBYTESRC_X43Y57/RIOI_I2GCLK_BOT1',),))
('LIOI3_X0Y183/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTESRC_X0Y181/LIOI_I2GCLK_TOP1',),))
('LIOI3_X0Y141/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTESRC_X0Y143/LIOI_I2GCLK_BOT1',),))
('RIOI_X43Y59/RIOI_I2GCLK_TOP0', (('RIOI_TBYTESRC_X43Y57/RIOI_I2GCLK_TOP1',),))
('LIOI3_X0Y191/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTESRC_X0Y193/LIOI_I2GCLK_BOT1',),))
('RIOI_X43Y29/RIOI_I2GCLK_TOP0', (('RIOI_TBYTESRC_X43Y31/RIOI_I2GCLK_BOT1',),))
('LIOI3_X0Y85/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTETERM_X0Y87/LIOI_I2GCLK_BOT1',),))
('LIOI3_TBYTESRC_X0Y143/LIOI_I2GCLK_TOP0', (('LIOI3_X0Y145/LIOI_I2GCLK_BOT1',), ('CMT_TOP_R_UPPER_B_X8Y135/CMT_PHASER_UP_DQS_TO_PHASER_D',), ('LIOI3_X0Y141/LIOI_I2GCLK_TOP1',)))
('LIOI3_X0Y29/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTESRC_X0Y31/LIOI_I2GCLK_BOT1',),))
('LIOI3_X0Y79/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTESRC_X0Y81/LIOI_I2GCLK_BOT1',),))
('LIOI3_X0Y89/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTETERM_X0Y87/LIOI_I2GCLK_TOP1',),))
('RIOI_TBYTESRC_X43Y19/RIOI_I2GCLK_TOP0', (('RIOI_X43Y21/RIOI_I2GCLK_BOT1',), ('RIOI_X43Y17/RIOI_I2GCLK_TOP1',)))
('RIOI_X43Y83/RIOI_I2GCLK_TOP0', (('RIOI_TBYTESRC_X43Y81/RIOI_I2GCLK_TOP1',),))
('RIOI_TBYTESRC_X43Y93/RIOI_I2GCLK_TOP0', (('RIOI_X43Y95/RIOI_I2GCLK_BOT1',), ('RIOI_X43Y91/RIOI_I2GCLK_TOP1',), ('CMT_TOP_L_UPPER_B_X108Y83/CMT_PHASER_UP_DQS_TO_PHASER_D',)))
('LIOI3_TBYTESRC_X0Y19/LIOI_I2GCLK_TOP0', (('LIOI3_X0Y21/LIOI_I2GCLK_BOT1',), ('LIOI3_X0Y17/LIOI_I2GCLK_TOP1',)))
('LIOI3_TBYTESRC_X0Y157/LIOI_I2GCLK_TOP0', (('CMT_TOP_R_LOWER_T_X8Y174/CMT_PHASER_DOWN_DQS_TO_PHASER_A',), ('LIOI3_X0Y155/LIOI_I2GCLK_TOP1',), ('LIOI3_X0Y159/LIOI_I2GCLK_BOT1',)))
('LIOI3_X0Y171/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTESRC_X0Y169/LIOI_I2GCLK_TOP1',),))
('LIOI3_X0Y35/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTETERM_X0Y37/LIOI_I2GCLK_BOT1',),))
('LIOI3_X0Y5/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTESRC_X0Y7/LIOI_I2GCLK_BOT1',),))
('LIOI3_X0Y109/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTESRC_X0Y107/LIOI_I2GCLK_TOP1',),))
('LIOI3_X0Y167/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTESRC_X0Y169/LIOI_I2GCLK_BOT1',),))
('LIOI3_X0Y117/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTESRC_X0Y119/LIOI_I2GCLK_BOT1',),))
('LIOI3_X0Y95/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTESRC_X0Y93/LIOI_I2GCLK_TOP1',),))
('RIOI_X43Y21/RIOI_I2GCLK_TOP0', (('RIOI_TBYTESRC_X43Y19/RIOI_I2GCLK_TOP1',),))
('LIOI3_X0Y111/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTETERM_X0Y113/LIOI_I2GCLK_BOT1',),))
('RIOI_X43Y5/RIOI_I2GCLK_TOP0', (('RIOI_TBYTESRC_X43Y7/RIOI_I2GCLK_BOT1',),))
('RIOI_X43Y85/RIOI_I2GCLK_TOP0', (('RIOI_TBYTETERM_X43Y87/RIOI_I2GCLK_BOT1',),))
('LIOI3_X0Y105/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTESRC_X0Y107/LIOI_I2GCLK_BOT1',),))
('LIOI3_X0Y145/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTESRC_X0Y143/LIOI_I2GCLK_TOP1',),))
('LIOI3_X0Y179/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTESRC_X0Y181/LIOI_I2GCLK_BOT1',),))
('RIOI_X43Y71/RIOI_I2GCLK_TOP0', (('RIOI_TBYTESRC_X43Y69/RIOI_I2GCLK_TOP1',),))
('LIOI3_TBYTESRC_X0Y31/LIOI_I2GCLK_TOP0', (('LIOI3_X0Y33/LIOI_I2GCLK_BOT1',), ('LIOI3_X0Y29/LIOI_I2GCLK_TOP1',)))
('RIOI_X43Y45/RIOI_I2GCLK_TOP0', (('RIOI_TBYTESRC_X43Y43/RIOI_I2GCLK_TOP1',),))
('RIOI_X43Y41/RIOI_I2GCLK_TOP0', (('RIOI_TBYTESRC_X43Y43/RIOI_I2GCLK_BOT1',),))
('RIOI_X43Y11/RIOI_I2GCLK_TOP0', (('RIOI_TBYTETERM_X43Y13/RIOI_I2GCLK_BOT1',),))
('LIOI3_X0Y55/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTESRC_X0Y57/LIOI_I2GCLK_BOT1',),))
('LIOI3_X0Y139/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTETERM_X0Y137/LIOI_I2GCLK_TOP1',),))
('LIOI3_X0Y71/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTESRC_X0Y69/LIOI_I2GCLK_TOP1',),))
('RIOI_X43Y89/RIOI_I2GCLK_TOP0', (('RIOI_TBYTETERM_X43Y87/RIOI_I2GCLK_TOP1',),))
('LIOI3_X0Y161/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTETERM_X0Y163/LIOI_I2GCLK_BOT1',),))
('LIOI3_X0Y133/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTESRC_X0Y131/LIOI_I2GCLK_TOP1',),))
('LIOI3_X0Y9/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTESRC_X0Y7/LIOI_I2GCLK_TOP1',),))
('RIOI_TBYTESRC_X43Y7/RIOI_I2GCLK_TOP0', (('CMT_TOP_L_LOWER_T_X108Y18/CMT_PHASER_DOWN_DQS_TO_PHASER_A',), ('RIOI_X43Y5/RIOI_I2GCLK_TOP1',), ('RIOI_X43Y9/RIOI_I2GCLK_BOT1',)))
('LIOI3_TBYTESRC_X0Y81/LIOI_I2GCLK_TOP0', (('LIOI3_X0Y83/LIOI_I2GCLK_BOT1',), ('LIOI3_X0Y79/LIOI_I2GCLK_TOP1',)))
('RIOI_X43Y95/RIOI_I2GCLK_TOP0', (('RIOI_TBYTESRC_X43Y93/RIOI_I2GCLK_TOP1',),))
('LIOI3_X0Y39/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTETERM_X0Y37/LIOI_I2GCLK_TOP1',),))
('LIOI3_TBYTESRC_X0Y181/LIOI_I2GCLK_TOP0', (('LIOI3_X0Y179/LIOI_I2GCLK_TOP1',), ('LIOI3_X0Y183/LIOI_I2GCLK_BOT1',)))
('LIOI3_X0Y155/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTESRC_X0Y157/LIOI_I2GCLK_BOT1',),))
('LIOI3_X0Y121/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTESRC_X0Y119/LIOI_I2GCLK_TOP1',),))
('LIOI3_X0Y33/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTESRC_X0Y31/LIOI_I2GCLK_TOP1',),))
('RIOI_X43Y9/RIOI_I2GCLK_TOP0', (('RIOI_TBYTESRC_X43Y7/RIOI_I2GCLK_TOP1',),))
[... cut ...]
('LIOI3_TBYTESRC_X0Y69/LIOI_I2GCLK_TOP0', (('LIOI3_X0Y67/LIOI_I2GCLK_TOP1',), ('LIOI3_X0Y71/LIOI_I2GCLK_BOT1',)))
('LIOI3_X0Y91/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTESRC_X0Y93/LIOI_I2GCLK_BOT1',),))
('LIOI3_X0Y115/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTETERM_X0Y113/LIOI_I2GCLK_TOP1',),))
('LIOI3_X0Y65/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTETERM_X0Y63/LIOI_I2GCLK_TOP1',),))
('LIOI3_X0Y83/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTESRC_X0Y81/LIOI_I2GCLK_TOP1',),))
('LIOI3_X0Y11/LIOI_I2GCLK_TOP0', (('LIOI3_TBYTETERM_X0Y13/LIOI_I2GCLK_BOT1',),))
```

mcmasterg, 2018-11-06:
@litghost thoughts?

litghost, 2018-11-06:
We just need to add the ignored wires file.  I'll put up a PR tomorrow.

PR #224 (2018-11-06): Add option to output ignored wires and add ignored wires for kintex7.
Fixes #223
closed 2018-11-06 by mithro