f4pga/prjxray #235: Parallelize full fuzzer build

JohnDMcMaster, 2018-11-07:
Refactor fuzzers/Makefile

Many of the fuzzer projects support make -j or at least should properly run with -j > 1. However, the top level makefile doesn't indicate any of the fuzzer dependencies and just makes all of them a dependency of all. So when we run make in that directory, it must be -j 1 to prevent parallelism dependency issues.

mcmasterg, 2018-11-08:
Actually this is parallelized in theory, but IIRC I ran into issues when trying -j. Lets verify it works correctly

mcmasterg, 2018-11-14:
Sounds like the real issue is that a few fuzzers use a large amount of memory (074?). We need to block parallelism on those

litghost, 2018-11-14:
Ya, 074 is itself parallelized, so I don't recommend parallelizing twice overtop it.

mcmasterg, 2018-11-15:
There are also mergedb issues if two fuzzers merge at the same time

mcmasterg, 2019-01-08:
In the sense that we understand the root cause, yes