f4pga/prjxray #241: New files need .gitignore

mithro, 2018-11-12:
Seems like we need a update to `.gitignore` maybe?
```
        005-tilegrid/tilegrid.json
        011-ffconfig/seg_clblx.segbits
        012-clbn5ffmux/seg_clblx.segbits
        013-clbncy0/seg_clblx.segbits
        014-ffsrcemux/seg_clblx.segbits
```

mcmasterg, 2018-11-12:
These sound like old files from build products getting moved into "build" dir. Can you confirm if you delete these they aren't regenerated?

mithro, 2018-11-15:
I'm now getting;
```
        010-lutinit/tmp.D7F8Nwz15v
        010-lutinit/tmp.dt3XOiePcF
        010-lutinit/tmp.nshzUDEgbj
        010-lutinit/tmp.vCAGjFwnmK
        011-ffconfig/tmp.EkWsoni9B0
        011-ffconfig/tmp.X7D5vT8yw3
        011-ffconfig/tmp.nU9vwzlQ81
        011-ffconfig/tmp.ompK5lXk1n
        012-clbn5ffmux/tmp.25OajO1nBz
        012-clbn5ffmux/tmp.ITVQLyagTz
        012-clbn5ffmux/tmp.aNTJ4mVBgW
        012-clbn5ffmux/tmp.zY10D4ZFiJ
        013-clbncy0/tmp.NU5HtQvKvD
        013-clbncy0/tmp.VAVc3lERhi
        013-clbncy0/tmp.eNvjL8WoK3
        013-clbncy0/tmp.n1nIJUrBCt
        014-ffsrcemux/tmp.YjuxjGJIqa
        014-ffsrcemux/tmp.nqcZnyCN4B
        014-ffsrcemux/tmp.tQmzI5QMot
        014-ffsrcemux/tmp.ugIqzjvJct
        015-clbnffmux/tmp.9Mxjsrs9TV
        015-clbnffmux/tmp.CYSyoeHnCC
        015-clbnffmux/tmp.YNxoUCgA1N
        015-clbnffmux/tmp.q8TDTWHE3e
        016-clbnoutmux/tmp.9aMaPXz8qY
        016-clbnoutmux/tmp.dLImLfEcHT
        016-clbnoutmux/tmp.oF5WBnY9iH
        016-clbnoutmux/tmp.rnNhZoD6oU
```

mithro, 2018-11-28:
As of 67e52e1fbae9916f7e3b23f1978d52c6dcb6500f (see below) I'm still seeing a bunch of tmp.XXX files turn up.
```
commit 67e52e1fbae9916f7e3b23f1978d52c6dcb6500f
Merge: 2297e0e 7f7a819
Author: John Mcmaster <JohnDMcMaster@gmail.com>
Date:   Thu Nov 15 11:45:46 2018 -0800

    Merge pull request #267 from mcmasterg/bram_segbits

    bram-data: fix build.db_bramx => build/segbits_bramx
```

These non-temp files have also turned up;
```
        fuzzers/010-lutinit/seg_clbll.db
        fuzzers/010-lutinit/seg_clblm.db
        fuzzers/050-intpips/build/
        fuzzers/055-gnd/seg_int_l.db
        fuzzers/055-gnd/seg_int_r.db
```

mithro, 2018-12-06:
Latest run as of afe50c68c464c0cd4a3fa92b6a07c9abbe41682f has the following;
```
        010-lutinit/seg_clbll.db
        010-lutinit/seg_clbll.segbits
        010-lutinit/seg_clblm.db
        050-intpips/build/       
        051-imuxlout/mask_bram_l.db
        051-imuxlout/mask_bram_r.db
        051-imuxlout/mask_clbll_l.db
        051-imuxlout/mask_clbll_r.db
        051-imuxlout/mask_clblm_l.db
        051-imuxlout/mask_clblm_r.db
        051-imuxlout/mask_dsp_l.db
        051-imuxlout/mask_dsp_r.db 
        052-clkin/mask_bram_l.db   
        052-clkin/mask_bram_r.db      
        052-clkin/mask_clbll_l.db     
        052-clkin/mask_clbll_r.db
        052-clkin/mask_clblm_l.db
        052-clkin/mask_clblm_r.db
        052-clkin/mask_dsp_l.db
        052-clkin/mask_dsp_r.db
        053-ctrlin/mask_bram_l.db
        053-ctrlin/mask_bram_r.db
        053-ctrlin/mask_clbll_l.db
        053-ctrlin/mask_clbll_r.db
        053-ctrlin/mask_clblm_l.db
        053-ctrlin/mask_clblm_r.db
        053-ctrlin/mask_dsp_l.db
        053-ctrlin/mask_dsp_r.db
        054-gfan/mask_bram_l.db
        054-gfan/mask_bram_r.db
        054-gfan/mask_clbll_l.db
        054-gfan/mask_clbll_r.db
        054-gfan/mask_clblm_l.db
        054-gfan/mask_clblm_r.db
        054-gfan/mask_dsp_l.db
        054-gfan/mask_dsp_r.db
        055-gnd/mask_bram_l.db
        055-gnd/mask_bram_r.db
        055-gnd/mask_clbll_l.db
        055-gnd/mask_clbll_r.db
        055-gnd/mask_clblm_l.db
        055-gnd/mask_clblm_r.db
        055-gnd/mask_dsp_l.db
        055-gnd/mask_dsp_r.db
        055-gnd/seg_int_l.db
        055-gnd/seg_int_r.db
        055-gnd/segbits_int_l.db
        055-gnd/segbits_int_r.db
        056-rempips/piplist.dcp
        056-rempips/piplist/
        056-rempips/pips_int_l.txt
        056-rempips/pips_int_r.txt
        056-rempips/todo.txt
        056-rempips/vivado.jou
        056-rempips/vivado.log
        057-bipips/mask_bram_l.db
        057-bipips/mask_bram_r.db
        057-bipips/mask_clbll_l.db
        057-bipips/mask_clbll_r.db
        057-bipips/mask_clblm_l.db
        057-bipips/mask_clblm_r.db
        057-bipips/mask_dsp_l.db
        057-bipips/mask_dsp_r.db
        058-hclkpips/mask_hclk_l.db
        058-hclkpips/mask_hclk_r.db
        058-hclkpips/segbits_hclk_l.db
        058-hclkpips/segbits_hclk_r.db
```

JohnDMcMaster, 2018-12-06:
Those are left over from previous rev. You just delete them