f4pga/prjxray #265: Duplicate tag INT_L.BYP_ALT0.BYP_BOUNCE_N3_3

mithro, 2018-11-15:
```
Reading mask bits from build/specimen_181/segdata_int_r.txt.
Reading mask bits from build/specimen_182/segdata_int_r.txt.
Reading mask bits from build/specimen_183/segdata_int_r.txt.
Reading mask bits from build/specimen_184/segdata_int_r.txt.
Reading mask bits from build/specimen_185/segdata_int_r.txt.
Reading mask bits from build/specimen_186/segdata_int_r.txt.
Reading mask bits from build/specimen_187/segdata_int_r.txt.
Reading mask bits from build/specimen_188/segdata_int_r.txt.
Reading mask bits from build/specimen_189/segdata_int_r.txt.
Reading mask bits from build/specimen_190/segdata_int_r.txt.
Reading mask bits from build/specimen_191/segdata_int_r.txt.
Reading mask bits from build/specimen_192/segdata_int_r.txt.
Reading mask bits from build/specimen_193/segdata_int_r.txt.
Reading mask bits from build/specimen_194/segdata_int_r.txt.
Reading mask bits from build/specimen_195/segdata_int_r.txt.
Reading mask bits from build/specimen_196/segdata_int_r.txt.
Reading mask bits from build/specimen_197/segdata_int_r.txt.
Reading mask bits from build/specimen_198/segdata_int_r.txt.
Reading mask bits from build/specimen_199/segdata_int_r.txt.
Reading mask bits from build/specimen_200/segdata_int_r.txt.
python3 /big/github/SymbiFlow/prjxray/utils/dbfixup.py --db-root build --clb-int
CLB INT mode: True
Segbit groups: 6
Segbit: checked 2 files w/ 3040 changed lines
Mask: checked files
make[2]: Leaving directory '/big/github/SymbiFlow/prjxray/fuzzers/050-intpips'
make pushdb
make[2]: Entering directory '/big/github/SymbiFlow/prjxray/fuzzers/050-intpips'
bash /big/github/SymbiFlow/prjxray/utils/mergedb.sh int_l build/segbits_int_l.db
+ test 2 = 2
+ test -e build/segbits_int_l.db
++ mktemp -p .
+ tmp1=./tmp.KHs3t2iWTU
++ mktemp -p .
+ tmp2=./tmp.YZDO4qWkIF
+ trap finish EXIT
+ db=/big/github/SymbiFlow/prjxray/database/artix7/segbits_int_l.db
+ case "$1" in
+ sed -e 's/^INT\./INT_L./'
+ touch /big/github/SymbiFlow/prjxray/database/artix7/segbits_int_l.db
+ sort -u ./tmp.KHs3t2iWTU /big/github/SymbiFlow/prjxray/database/artix7/segbits_int_l.db
+ grep -v '<.*>'
+ python3 /big/github/SymbiFlow/prjxray/utils/parsedb.py --strict ./tmp.YZDO4qWkIF /big/github/SymbiFlow/prjxray/database/artix7/segbits_int_l.db
Original line: INT_L.BYP_ALT0.BYP_BOUNCE_N3_3 21_07 24_07
New line: INT_L.BYP_ALT0.BYP_BOUNCE_N3_3 !22_07 !23_07 !25_07 21_07 24_07
Traceback (most recent call last):
  File "/big/github/SymbiFlow/prjxray/utils/parsedb.py", line 60, in <module>
    main()
  File "/big/github/SymbiFlow/prjxray/utils/parsedb.py", line 56, in main
    run(args.fin, args.fout, strict=args.strict, verbose=args.verbose)
  File "/big/github/SymbiFlow/prjxray/utils/parsedb.py", line 24, in run
    assert 0, "strict: got duplicate tag %s" % (tag, )
AssertionError: strict: got duplicate tag INT_L.BYP_ALT0.BYP_BOUNCE_N3_3
+ finish
+ echo 'Cleaning up temp files'
Cleaning up temp files
+ rm -f ./tmp.KHs3t2iWTU
+ rm -f ./tmp.YZDO4qWkIF
Makefile:16: recipe for target 'pushdb' failed
make[2]: *** [pushdb] Error 1
make[2]: Leaving directory '/big/github/SymbiFlow/prjxray/fuzzers/050-intpips'
Makefile:29: recipe for target 'run' failed
make[1]: *** [run] Error 2
make[1]: Leaving directory '/big/github/SymbiFlow/prjxray/fuzzers/050-intpips'
Makefile:23: recipe for target '050-intpips/run.ok' failed
make: *** [050-intpips/run.ok] Error 2
[UPDATE PRJXRAY DATABASE] tansell@instance-1:~/github/SymbiFlow/prjxray/fuzzers$
```

JohnDMcMaster, 2018-11-15:
Are you sure your workspace was clean before running? This implies to me /big/github/SymbiFlow/prjxray/database/artix7/segbits_int_l.db had a stale entry in it. More specifically, this implies it had an entry from before I merged https://github.com/SymbiFlow/prjxray/pull/262 which was causing dbfixup not to run in some instances

mithro, 2018-11-15:
I just updated and restarted the fuzzers, closing for now.

closed 2018-11-15 by mithro