f4pga/prjxray #266: No rule to make target 'build/segbits_bramx.rdb', needed by 'build.db_bramx.db'

mithro, 2018-11-15:
```
+ tmp2=./tmp.6Ai9D6PF6R
+ trap finish EXIT
+ db=/big/github/SymbiFlow/prjxray/database/artix7/segbits_clblm_r.db
+ case "$1" in
+ sed -e 's/^CLB\.SLICE_X0\./CLBLM_R.SLICEM_X0./' -e 's/^CLB\.SLICE_X1\./CLBLM_R.SLICEL_X1./'
+ touch /big/github/SymbiFlow/prjxray/database/artix7/segbits_clblm_r.db
+ sort -u ./tmp.lnbEbuc9N8 /big/github/SymbiFlow/prjxray/database/artix7/segbits_clblm_r.db
+ grep -v '<.*>'
+ python3 /big/github/SymbiFlow/prjxray/utils/parsedb.py --strict ./tmp.6Ai9D6PF6R /big/github/SymbiFlow/prjxray/database/artix7/segbits_clblm_r.db
+ finish
+ echo 'Cleaning up temp files'
Cleaning up temp files
+ rm -f ./tmp.lnbEbuc9N8
+ rm -f ./tmp.6Ai9D6PF6R
make[2]: Leaving directory '/big/github/SymbiFlow/prjxray/fuzzers/019-ndi1mux'
touch run.ok
make[1]: Leaving directory '/big/github/SymbiFlow/prjxray/fuzzers/019-ndi1mux'
make -C 025-bram-config run
make[1]: Entering directory '/big/github/SymbiFlow/prjxray/fuzzers/025-bram-config'
make clean
make[2]: Entering directory '/big/github/SymbiFlow/prjxray/fuzzers/025-bram-config'
rm -rf build run.ok
make[2]: Leaving directory '/big/github/SymbiFlow/prjxray/fuzzers/025-bram-config'
make database
make[2]: Entering directory '/big/github/SymbiFlow/prjxray/fuzzers/025-bram-config'
make[2]: *** No rule to make target 'build/segbits_bramx.rdb', needed by 'build.db_bramx.db'.  Stop.
make[2]: Leaving directory '/big/github/SymbiFlow/prjxray/fuzzers/025-bram-config'
../fuzzer.mk:13: recipe for target 'run' failed
make[1]: *** [run] Error 2
make[1]: Leaving directory '/big/github/SymbiFlow/prjxray/fuzzers/025-bram-config'
Makefile:21: recipe for target '025-bram-config/run.ok' failed
make: *** [025-bram-config/run.ok] Error 2
[UPDATE PRJXRAY DATABASE] tansell@instance-1:~/github/SymbiFlow/prjxray/fuzzers$
```

JohnDMcMaster, 2018-11-15:
tentative PR: https://github.com/SymbiFlow/prjxray/pull/267

Looks like only 025 was affected. I'm running local builds for 025/026 right now to be sure. Hopefully they will complete by the time travis oks the PR

JohnDMcMaster, 2018-11-15:
025 including push looks okay now. 026 possibly has a data loss issue, although the build does complete

JohnDMcMaster, 2018-11-15:
Data loss was caused by my local DB not having BRAM memory region. Think its fine

mithro, 2018-12-07:
@JohnDMcMaster Is this fixed now?

mcmasterg, 2018-12-10:
Yes

closed 2018-12-10 by mcmasterg