f4pga/prjxray #272: 056-rempips gets stuck in an infinite loop

mithro, 2018-11-17:
It looks like `056-rempips` is currently broken. It gets stuck in an infinite loop. I think the issue is show below;
```
Loading data files...                                                            
Loading site data...          
Loading route data...                                                            
Processing options...     
Creating bitmap...                                                               
Creating bitstream...                     
Writing bitstream ./design.bit...                                                
INFO: [Vivado 12-1842] Bitgen Completed Successfully.     
INFO: [Project 1-120] WebTalk data collection is mandatory when using a WebPACK part without a full Vivado license. To see the specific WebTalk data collected for your design, open the usage_statistics_webtalk.html or usage_statistics_webtalk.xml file in the implementation directory.
INFO: [Common 17-186] '/big/github/SymbiFlow/prjxray/fuzzers/056-rempips/specimen_001/design/usage_statistics_webtalk.xml' has been successfully sent to Xilinx on Sat Nov 17 19:43:16 2018. For additional details about this file, please refer to the WebTalk help file at /opt/Xilinx/Vivado/2017.2/doc/webtalk_introduction.html.
9 Infos, 1 Warnings, 0 Critical Warnings and 0 Errors encountered.               
write_bitstream completed successfully
write_bitstream: Time (s): cpu = 00:00:08 ; elapsed = 00:00:08 . Memory (MB): peak = 2380.887 ; gain = 181.352 ; free physical = 287884 ; free virtual = 356802
# write_txtdata design.txt                
Writing design.txt.                       
Dumping pips.                             
INFO: [Common 17-206] Exiting Vivado at Sat Nov 17 19:43:17 2018...
+ /big/github/SymbiFlow/prjxray/build/tools/bitread --part_file /big/github/SymbiFlow/prjxray/database/artix7/xc7a50tfgg484-1.yaml -F 0x00000000:0xffffffff -o design.bits -z -y design.bit
Bitstream size: 2298099 bytes                                                    
Config size: 574487 words
Number of configuration frames: 5408                                             
DONE                                                                             
+ python3 ../generate.py                                                                                                                                                     
Loading bits from design.bits.                                                   
Loading todo from ../todo.txt.
Loading tags from design.txt.
Compiling segment data.                   
Traceback (most recent call last):        
  File "../generate.py", line 65, in <module>
    segmk.write()                         
  File "/big/github/SymbiFlow/prjxray/prjxray/segmaker.py", line 334, in write   
    ]) != 0, "Didn't  generate any segments"
AssertionError: Didn't  generate any segments                                    
Makefile:16: recipe for target 'specimen_001/OK' failed                          
make[2]: *** [specimen_001/OK] Error 1                                                                                                                                      
make[2]: Leaving directory '/big/github/SymbiFlow/prjxray/fuzzers/056-rempips'   
+ make clean
make[2]: Entering directory '/big/github/SymbiFlow/prjxray/fuzzers/056-rempips'
rm -rf .Xil/ .cache/ filtered_seg_int_[lr].db run.ok
rm -rf todo.txt vivado* piplist/ piplist.dcp pattern_[lr].txt pips_int_[lr].txt
rm -rf specimen_[0-9][0-9][0-9]/ seg_int_[lr].db
make[2]: Leaving directory '/big/github/SymbiFlow/prjxray/fuzzers/056-rempips'
+ make todo.txt                                                                  
make[2]: Entering directory '/big/github/SymbiFlow/prjxray/fuzzers/056-rempips'
```

JohnDMcMaster, 2018-11-27:
Can you:
-make todo.txt
-Give me todo.txt?

mithro, 2018-11-27:
`more todo.txt` gives;
```
INT_R.BYP_ALT6.LOGIC_OUTS7
```

mithro, 2018-11-27:
`more todo.txt` for the Kintex-7 gives;
```
INT_L.BYP_ALT7.LOGIC_OUTS_L3
INT_R.BYP_ALT6.LOGIC_OUTS7
```

JohnDMcMaster, 2018-11-28:
Something to do with route_via:

```
Routing net mynet_0:
  CLBLL_L_X2Y146/CLBLL_L_A -> CLBLM_R_X7Y115/CLBLM_LOGIC_OUTS7: no route found - assuming direct PIP
  CLBLM_R_X7Y115/CLBLM_LOGIC_OUTS7 -> INT_R_X7Y115/BYP_ALT6: no route found - assuming direct PIP
  INT_R_X7Y115/BYP_ALT6 -> CLBLL_L_X2Y146/CLBLL_L_A5: INT_R_X7Y115/BYP_ALT6 INT_R_X7Y115/BYP_BOUNCE6 INT_R_X7Y116/IMUX40 CLBLM_R_X7Y116/CLBLM_M_D1 CLBLM_R_X7Y116/CLBLM_M_D CLBLM_R_X7Y116/CLBLM_LOGIC_OUTS15 INT_R_X7Y116/SR1BEG_S0 INT_R_X7Y134/LV18 INT_R_X7Y146/LVB12 INT_R_X7Y146/WW4BEG2 INT_R_X3Y146/WL1BEG0 INT_L_X2Y146/IMUX_L9 CLBLL_L_X2Y146/CLBLL_L_A5
```

It fails to route the net. generate.tcl subsequently route_design, causing it to get routed into whatever Vivado feels like. It correctly writes this unexpected metadata, which doesn't line up with the expected route in todo.txt. Since the expected route doesn't exist, its ignored. Since no expected routes were found, no tags are added. This eventually cases segmaker to fail

DB is generally finishy. Take a look at this old DB entry:
```
database.1/kintex7/segbits_int_l.db:699:INT_L.FAN_ALT5.BYP_BOUNCE1 !22_40 !23_40 !25_40 20_40 24_40
database.1/kintex7/segbits_int_r.db:699:INT_R.FAN_ALT5.BYP_BOUNCE1 !22_40 !23_40 !25_40 20_40 24_40
```

However my local DB has:

```
int_l.db: FAN_ALT5.BYP_BOUNCE1 !22_40 !23_40 !25_40 20_40 24_40
int_r.db: FAN_ALT5.BYP_BOUNCE1 20_40 24_40
```
Somehow l and r are out of sync. This is somewhat expected as they run independently in this fuzzer, but its unclear why the mask is missing on the _r entry.

JohnDMcMaster, 2018-11-28:
Filed https://github.com/SymbiFlow/prjxray/issues/276

I'm not sure this will solve the core issue, but suggest fixing it first and see if it improves things. I can manually do dbfixup on the root db without re-running everything

JohnDMcMaster, 2018-11-28:
Fixed the 276, doesn't effect

Confirmed its a routing issue:
```
set_property FIXED_ROUTE $fixed_route $mynet
ERROR: [Designutils 20-941] Did not find node resource, CLBLM_R_X7Y115/CLBLM_LOGIC_OUTS7, downhill from node, CLBLL_L_X2Y146/CLBLL_L_A.
Downhill node choices include:  CLBLL_L_X2Y146/CLBLL_LOGIC_OUTS8 CLBLL_L_X2Y146/CLBLL_L_AMUX .
```

Since we had -quiet, the error had been suppressed

JohnDMcMaster, 2018-11-28:
So...LUTs are LOC'd and BEL'd:
```
set_property -dict "LOC $driver_site BEL A6LUT" $mylut
```
This is problematic since CLBLM_LOGIC_OUTS7 is for DQ and not routable from A6LUT. I'm guessing that although 050 does 200 runs, somehow this got missed. Maybe we should add more FFs? With the recent BRAM changes, PnR just happened to change to eliminate this case.

Finally, this implies rempips was only meant to target a few very specific PIPs? Or maybe it was usually able to figure out a route through the carry chain or similar, but failed in this particular case.

JohnDMcMaster, 2018-11-28:
Let it run last night with the FF patch. Took a while (415m) but it did complete. Iteration log:

```
wc -l todo/*all.txt |sort -r -n
 5143 total
  276 todo/4_all.txt
  276 todo/3_all.txt
  276 todo/2_all.txt
  276 todo/1_all.txt
  266 todo/5_all.txt
  256 todo/6_all.txt
  246 todo/7_all.txt
  236 todo/8_all.txt
  226 todo/9_all.txt
  216 todo/10_all.txt
  206 todo/11_all.txt
  196 todo/12_all.txt
  187 todo/13_all.txt
  177 todo/14_all.txt
  167 todo/15_all.txt
  157 todo/16_all.txt
  147 todo/18_all.txt
  147 todo/17_all.txt
  137 todo/19_all.txt
  127 todo/20_all.txt
  117 todo/22_all.txt
  117 todo/21_all.txt
  107 todo/23_all.txt
   97 todo/24_all.txt
   87 todo/25_all.txt
   77 todo/26_all.txt
   67 todo/27_all.txt
   57 todo/29_all.txt
   57 todo/28_all.txt
   47 todo/30_all.txt
   37 todo/31_all.txt
   27 todo/33_all.txt
   27 todo/32_all.txt
   17 todo/34_all.txt
    7 todo/35_all.txt
```

closed 2018-12-11 by mcmasterg