f4pga/prjxray #293: 0 candidates for SRL and SMALL for {A,B.C}LUT (in CLBM)

mithro, 2018-12-07:
The [`018-clbram` fuzzer](https://github.com/SymbiFlow/prjxray/tree/master/fuzzers/018-clbram) seems unable to find candidates for the following;
```
::::::::::::::
build/segbits_clbx.db
::::::::::::::
CLB.SLICE_X0.ALUT.RAM 31_16
CLB.SLICE_X0.ALUT.SMALL <0 candidates>
CLB.SLICE_X0.ALUT.SRL <0 candidates>
CLB.SLICE_X0.BLUT.RAM 31_17
CLB.SLICE_X0.BLUT.SMALL <0 candidates>
CLB.SLICE_X0.BLUT.SRL <0 candidates>
CLB.SLICE_X0.CLUT.RAM 31_46
CLB.SLICE_X0.CLUT.SMALL <0 candidates>
CLB.SLICE_X0.CLUT.SRL <0 candidates>
CLB.SLICE_X0.DLUT.RAM 31_47
CLB.SLICE_X0.DLUT.SMALL 01_59
CLB.SLICE_X0.DLUT.SRL 30_47
CLB.SLICE_X0.WA7USED 00_40
CLB.SLICE_X0.WA8USED 01_27
CLB.SLICE_X0.WEMUX.CE 01_23
```

Previously these seem to have been;
```
CLBLM_R.SLICEM_X0.ALUT.SMALL 00_04
CLBLM_R.SLICEM_X0.ALUT.SRL 30_16
CLBLM_R.SLICEM_X0.BLUT.SMALL 00_24
CLBLM_R.SLICEM_X0.BLUT.SRL 30_17
CLBLM_R.SLICEM_X0.CLUT.SMALL 00_28
CLBLM_R.SLICEM_X0.CLUT.SRL 30_46
```

All these bits seem to be unused by anything else in the file?

mithro, 2018-12-07:
Strangely the `DLUT.SMALL` and `DLUT.SRL` are found.

mcmasterg, 2018-12-11:
FYI this is breaking prjxray-quick

closed 2018-12-11 by mcmasterg