f4pga/prjxray #315: checkdb failures

mcmasterg, 2018-12-11:
When the database was restructured from tiles to segments, we talked about adding a tool to detect collisions but it was only added recently.  Now that I'm checking we are finding failures.

Jenkins full run 3 (last good build I have) output:

```
$ python3 utils/checkdb.py  --db-root database.f-3/artix7/
Checking individual .db...
segbits_*.db: 12 okay
mask_*.db: 6 okay
Checking aggregate dir...
ERROR: 11 collisions
  00020181_096_07: had CLBLM_R_X3Y147.CLBLM_R.SLICEM_X0.CEUSEDMUX, got INT_R_X3Y147.INT_R.FAN_ALT7.GFAN1
  00020199_096_00: had CLBLM_R_X3Y147.CLBLM_R.SLICEM_X0.WEMUX.CE, got INT_R_X3Y147.INT_R.FAN_ALT7.BYP_BOUNCE0
  00020199_096_31: had CLBLM_R_X3Y147.CLBLM_R.SLICEL_X1.D5FFMUX.IN_B, got INT_R_X3Y147.INT_R.BYP_ALT7.BYP_BOUNCE2
  0002019e_095_02: had CLBLM_R_X3Y147.CLBLM_R.SLICEM_X0.AFFMUX.AX, got INT_R_X3Y147.INT_R.BYP_ALT1.VCC_WIRE
  0002019e_095_25: had CLBLM_R_X3Y147.CLBLM_R.SLICEM_X0.BFFMUX.BX, got INT_R_X3Y147.INT_R.BYP_ALT1.VCC_WIRE
  0002019e_096_05: had CLBLM_R_X3Y147.CLBLM_R.SLICEM_X0.CFFMUX.CX, got INT_R_X3Y147.INT_R.BYP_ALT1.VCC_WIRE
  0002019e_096_28: had CLBLM_R_X3Y147.CLBLM_R.SLICEM_X0.DFFMUX.CY, got INT_R_X3Y147.INT_R.BYP_ALT1.VCC_WIRE
  0002019f_095_02: had CLBLM_R_X3Y147.CLBLM_R.SLICEL_X1.AFFMUX.AX, got INT_R_X3Y147.INT_R.BYP_ALT0.VCC_WIRE
  0002019f_095_26: had CLBLM_R_X3Y147.CLBLM_R.SLICEL_X1.BFFMUX.BX, got INT_R_X3Y147.INT_R.BYP_ALT0.VCC_WIRE
  0002019f_096_05: had CLBLM_R_X3Y147.CLBLM_R.SLICEL_X1.CFFMUX.CX, got INT_R_X3Y147.INT_R.BYP_ALT0.VCC_WIRE
  0002019f_096_30: had CLBLM_R_X3Y147.CLBLM_R.SLICEL_X1.DFFMUX.CY, got INT_R_X3Y147.INT_R.BYP_ALT0.VCC_WIRE
Traceback (most recent call last):
  File "utils/checkdb.py", line 136, in <module>
    main()
  File "utils/checkdb.py", line 132, in main
    run(args.db_root, strict=args.strict, verbose=args.verbose)
  File "utils/checkdb.py", line 118, in run
    check_tile_overlap(db, db_root, strict=strict, verbose=verbose)
  File "utils/checkdb.py", line 94, in check_tile_overlap
    raise ValueError("%s collisions" % len(collisions))
ValueError: 11 collisions
```

mcmasterg, 2018-12-19:
This is still pending a full good DB build

mithro, 2018-12-20:
What database revision is this being run on?

mcmasterg, 2018-12-20:
Above output is from the last full jenkins run (3). However, there hasn't been a full successful build since then

mithro, 2018-12-20:
What does it produce on the current head at https://github.com/SymbiFlow/prjxray-db ?

mcmasterg, 2018-12-20:
Fails before even making it to aggregate db check due to renamed tags

```
mcmaster@xc2064:~/prjxray$ python3 utils/checkdb.py  --db-root database.latest/artix7/
Checking individual .db...
Traceback (most recent call last):
  File "utils/checkdb.py", line 136, in <module>
    main()
  File "utils/checkdb.py", line 132, in main
    run(args.db_root, strict=args.strict, verbose=args.verbose)
  File "utils/checkdb.py", line 103, in run
    parsedb_all(db_root, verbose=verbose)
  File "utils/checkdb.py", line 45, in parsedb_all
    parsedb.run(bit_fn, fnout=None, strict=True, verbose=verbose)
  File "/home/mcmaster/prjxray/utils/parsedb.py", line 27, in run
    bits, tag, bitss[bits])
AssertionError: strict: got duplicate bits frozenset({'30_09'}): CLBLL_R.SLICEL_X0.A5FFMUX.IN_A CLBLL_R.SLICEL_X0.A5FF.MUX.A
```

mcmasterg, 2018-12-20:
Actually you have a newer db than I thought. But basically the same issue as jenkins

```
$ python3 utils/checkdb.py  --db-root database/artix7/
Checking individual .db...
segbits_*.db: 12 okay
mask_*.db: 6 okay
Checking aggregate dir...
ERROR: 8 collisions
  0040099e_000_02: had INT_R_X19Y0.INT_R.BYP_ALT1.VCC_WIRE, got CLBLL_R_X19Y0.CLBLL_R.SLICEL_X0.AFFMUX.AX
  0040099e_000_25: had INT_R_X19Y0.INT_R.BYP_ALT1.VCC_WIRE, got CLBLL_R_X19Y0.CLBLL_R.SLICEL_X0.BFFMUX.BX
  0040099e_001_05: had INT_R_X19Y0.INT_R.BYP_ALT1.VCC_WIRE, got CLBLL_R_X19Y0.CLBLL_R.SLICEL_X0.CFFMUX.CX
  0040099e_001_28: had INT_R_X19Y0.INT_R.BYP_ALT1.VCC_WIRE, got CLBLL_R_X19Y0.CLBLL_R.SLICEL_X0.DFFMUX.CY
  0040099f_000_02: had INT_R_X19Y0.INT_R.BYP_ALT0.VCC_WIRE, got CLBLL_R_X19Y0.CLBLL_R.SLICEL_X1.AFFMUX.AX
  0040099f_000_26: had INT_R_X19Y0.INT_R.BYP_ALT0.VCC_WIRE, got CLBLL_R_X19Y0.CLBLL_R.SLICEL_X1.BFFMUX.BX
  0040099f_001_05: had INT_R_X19Y0.INT_R.BYP_ALT0.VCC_WIRE, got CLBLL_R_X19Y0.CLBLL_R.SLICEL_X1.CFFMUX.CX
  0040099f_001_30: had INT_R_X19Y0.INT_R.BYP_ALT0.VCC_WIRE, got CLBLL_R_X19Y0.CLBLL_R.SLICEL_X1.DFFMUX.CY
Traceback (most recent call last):
  File "utils/checkdb.py", line 136, in <module>
    main()
  File "utils/checkdb.py", line 132, in main
    run(args.db_root, strict=args.strict, verbose=args.verbose)
  File "utils/checkdb.py", line 118, in run
    check_tile_overlap(db, db_root, strict=strict, verbose=verbose)
  File "utils/checkdb.py", line 94, in check_tile_overlap
    raise ValueError("%s collisions" % len(collisions))
ValueError: 8 collisions
```

mithro, 2018-12-20:
FYI - The output of this tool doesn't seem to be deterministic?

mithro, 2018-12-20:
```
(env) tansell@tansell:~/github/SymbiFlow/prjxray$ ./utils/checkdb.py --db-root ../prjxray-db/artix7
Checking individual .db...
segbits_*.db: 12 okay
mask_*.db: 6 okay
Checking aggregate dir...
ERROR: 8 collisions
  0002069e_014_02: had CLBLL_R_X13Y107.CLBLL_R.SLICEL_X0.AFFMUX.AX, got INT_R_X13Y107.INT_R.BYP_ALT1.VCC_WIRE
  0002069e_014_25: had CLBLL_R_X13Y107.CLBLL_R.SLICEL_X0.BFFMUX.BX, got INT_R_X13Y107.INT_R.BYP_ALT1.VCC_WIRE
  0002069e_015_05: had CLBLL_R_X13Y107.CLBLL_R.SLICEL_X0.CFFMUX.CX, got INT_R_X13Y107.INT_R.BYP_ALT1.VCC_WIRE
  0002069e_015_28: had CLBLL_R_X13Y107.CLBLL_R.SLICEL_X0.DFFMUX.CY, got INT_R_X13Y107.INT_R.BYP_ALT1.VCC_WIRE
  0002069f_014_02: had CLBLL_R_X13Y107.CLBLL_R.SLICEL_X1.AFFMUX.AX, got INT_R_X13Y107.INT_R.BYP_ALT0.VCC_WIRE
  0002069f_014_26: had CLBLL_R_X13Y107.CLBLL_R.SLICEL_X1.BFFMUX.BX, got INT_R_X13Y107.INT_R.BYP_ALT0.VCC_WIRE
  0002069f_015_05: had CLBLL_R_X13Y107.CLBLL_R.SLICEL_X1.CFFMUX.CX, got INT_R_X13Y107.INT_R.BYP_ALT0.VCC_WIRE
  0002069f_015_30: had CLBLL_R_X13Y107.CLBLL_R.SLICEL_X1.DFFMUX.CY, got INT_R_X13Y107.INT_R.BYP_ALT0.VCC_WIRE
Traceback (most recent call last):
  File "./utils/checkdb.py", line 136, in <module>
    main()
  File "./utils/checkdb.py", line 132, in main
    run(args.db_root, strict=args.strict, verbose=args.verbose)
  File "./utils/checkdb.py", line 118, in run
    check_tile_overlap(db, db_root, strict=strict, verbose=verbose)
  File "./utils/checkdb.py", line 94, in check_tile_overlap
    raise ValueError("%s collisions" % len(collisions))
ValueError: 8 collisions
(env) tansell@tansell:~/github/SymbiFlow/prjxray$ ./utils/checkdb.py --db-root ../prjxray-db/artix7
Checking individual .db...
segbits_*.db: 12 okay
mask_*.db: 6 okay
Checking aggregate dir...

ERROR: 8 collisions
  00400c1e_087_02: had INT_L_X24Y43.INT_L.BYP_ALT1.VCC_WIRE, got CLBLL_L_X24Y43.CLBLL_L.SLICEL_X0.AFFMUX.AX
  00400c1e_087_25: had INT_L_X24Y43.INT_L.BYP_ALT1.VCC_WIRE, got CLBLL_L_X24Y43.CLBLL_L.SLICEL_X0.BFFMUX.BX
  00400c1e_088_05: had INT_L_X24Y43.INT_L.BYP_ALT1.VCC_WIRE, got CLBLL_L_X24Y43.CLBLL_L.SLICEL_X0.CFFMUX.CX
  00400c1e_088_28: had INT_L_X24Y43.INT_L.BYP_ALT1.VCC_WIRE, got CLBLL_L_X24Y43.CLBLL_L.SLICEL_X0.DFFMUX.CY
  00400c1f_087_02: had INT_L_X24Y43.INT_L.BYP_ALT0.VCC_WIRE, got CLBLL_L_X24Y43.CLBLL_L.SLICEL_X1.AFFMUX.AX
  00400c1f_087_26: had INT_L_X24Y43.INT_L.BYP_ALT0.VCC_WIRE, got CLBLL_L_X24Y43.CLBLL_L.SLICEL_X1.BFFMUX.BX
  00400c1f_088_05: had INT_L_X24Y43.INT_L.BYP_ALT0.VCC_WIRE, got CLBLL_L_X24Y43.CLBLL_L.SLICEL_X1.CFFMUX.CX
  00400c1f_088_30: had INT_L_X24Y43.INT_L.BYP_ALT0.VCC_WIRE, got CLBLL_L_X24Y43.CLBLL_L.SLICEL_X1.DFFMUX.CY
Traceback (most recent call last):
  File "./utils/checkdb.py", line 136, in <module>
    main()
  File "./utils/checkdb.py", line 132, in main
    run(args.db_root, strict=args.strict, verbose=args.verbose)
  File "./utils/checkdb.py", line 118, in run
    check_tile_overlap(db, db_root, strict=strict, verbose=verbose)
  File "./utils/checkdb.py", line 94, in check_tile_overlap
    raise ValueError("%s collisions" % len(collisions))
ValueError: 8 collisions
                                          
```

mithro, 2018-12-20:
```
mask_*.db: 6 okay
Checking aggregate dir...
ERROR: 4 collisions
  0040099e_090_05: had INT_R_X19Y44.INT_R.BYP_ALT1.VCC_WIRE, got CLBLL_R_X19Y44.CLBLL_R.SLICEL_X0.CFFMUX.CX
  0040099e_090_28: had INT_R_X19Y44.INT_R.BYP_ALT1.VCC_WIRE, got CLBLL_R_X19Y44.CLBLL_R.SLICEL_X0.DFFMUX.CY
  0040099f_090_05: had INT_R_X19Y44.INT_R.BYP_ALT0.VCC_WIRE, got CLBLL_R_X19Y44.CLBLL_R.SLICEL_X1.CFFMUX.CX
  0040099f_090_30: had INT_R_X19Y44.INT_R.BYP_ALT0.VCC_WIRE, got CLBLL_R_X19Y44.CLBLL_R.SLICEL_X1.DFFMUX.CY
ERROR: 5 collisions
  00001081_100_11: had INT_R_X33Y99.INT_R.FAN_ALT5.GFAN1, got CLBLM_R_X33Y99.CLBLM_R.SLICEM_X0.CLUT.DI1MUX.CI
  0000109e_100_05: had INT_R_X33Y99.INT_R.BYP_ALT1.VCC_WIRE, got CLBLM_R_X33Y99.CLBLM_R.SLICEM_X0.CFFMUX.CX
  0000109e_100_28: had INT_R_X33Y99.INT_R.BYP_ALT1.VCC_WIRE, got CLBLM_R_X33Y99.CLBLM_R.SLICEM_X0.DFFMUX.CY
  0000109f_100_05: had INT_R_X33Y99.INT_R.BYP_ALT0.VCC_WIRE, got CLBLM_R_X33Y99.CLBLM_R.SLICEL_X1.CFFMUX.CX
  0000109f_100_30: had INT_R_X33Y99.INT_R.BYP_ALT0.VCC_WIRE, got CLBLM_R_X33Y99.CLBLM_R.SLICEL_X1.DFFMUX.CY
ERROR: 4 collisions
  0002011e_029_05: had INT_L_X2Y114.INT_L.BYP_ALT1.VCC_WIRE, got CLBLL_L_X2Y114.CLBLL_L.SLICEL_X0.CFFMUX.CX
  0002011e_029_28: had INT_L_X2Y114.INT_L.BYP_ALT1.VCC_WIRE, got CLBLL_L_X2Y114.CLBLL_L.SLICEL_X0.DFFMUX.CY
  0002011f_029_05: had INT_L_X2Y114.INT_L.BYP_ALT0.VCC_WIRE, got CLBLL_L_X2Y114.CLBLL_L.SLICEL_X1.CFFMUX.CX
  0002011f_029_30: had INT_L_X2Y114.INT_L.BYP_ALT0.VCC_WIRE, got CLBLL_L_X2Y114.CLBLL_L.SLICEL_X1.DFFMUX.CY
ERROR: 5 collisions
  00020281_005_11: had INT_R_X5Y102.INT_R.FAN_ALT5.GFAN1, got CLBLM_R_X5Y102.CLBLM_R.SLICEM_X0.CLUT.DI1MUX.CI
  0002029e_005_05: had INT_R_X5Y102.INT_R.BYP_ALT1.VCC_WIRE, got CLBLM_R_X5Y102.CLBLM_R.SLICEM_X0.CFFMUX.CX
  0002029e_005_28: had INT_R_X5Y102.INT_R.BYP_ALT1.VCC_WIRE, got CLBLM_R_X5Y102.CLBLM_R.SLICEM_X0.DFFMUX.CY
  0002029f_005_05: had INT_R_X5Y102.INT_R.BYP_ALT0.VCC_WIRE, got CLBLM_R_X5Y102.CLBLM_R.SLICEL_X1.CFFMUX.CX
  0002029f_005_30: had INT_R_X5Y102.INT_R.BYP_ALT0.VCC_WIRE, got CLBLM_R_X5Y102.CLBLM_R.SLICEL_X1.DFFMUX.CY
ERROR: 4 collisions
  0000101e_068_05: had CLBLM_L_X32Y83.CLBLM_L.SLICEM_X0.CFFMUX.CX, got INT_L_X32Y83.INT_L.BYP_ALT1.VCC_WIRE
  0000101e_068_28: had CLBLM_L_X32Y83.CLBLM_L.SLICEM_X0.DFFMUX.CY, got INT_L_X32Y83.INT_L.BYP_ALT1.VCC_WIRE
  0000101f_068_05: had CLBLM_L_X32Y83.CLBLM_L.SLICEL_X1.CFFMUX.CX, got INT_L_X32Y83.INT_L.BYP_ALT0.VCC_WIRE
  0000101f_068_30: had CLBLM_L_X32Y83.CLBLM_L.SLICEL_X1.DFFMUX.CY, got INT_L_X32Y83.INT_L.BYP_ALT0.VCC_WIRE
ERROR: 5 collisions
  00000c81_092_11: had INT_R_X25Y95.INT_R.FAN_ALT5.GFAN1, got CLBLM_R_X25Y95.CLBLM_R.SLICEM_X0.CLUT.DI1MUX.CI
  00000c9e_092_05: had INT_R_X25Y95.INT_R.BYP_ALT1.VCC_WIRE, got CLBLM_R_X25Y95.CLBLM_R.SLICEM_X0.CFFMUX.CX
  00000c9e_092_28: had INT_R_X25Y95.INT_R.BYP_ALT1.VCC_WIRE, got CLBLM_R_X25Y95.CLBLM_R.SLICEM_X0.DFFMUX.CY
  00000c9f_092_05: had INT_R_X25Y95.INT_R.BYP_ALT0.VCC_WIRE, got CLBLM_R_X25Y95.CLBLM_R.SLICEL_X1.CFFMUX.CX
  00000c9f_092_30: had INT_R_X25Y95.INT_R.BYP_ALT0.VCC_WIRE, got CLBLM_R_X25Y95.CLBLM_R.SLICEL_X1.DFFMUX.CY
ERROR: 4 collisions
  00400a9e_015_05: had INT_R_X21Y7.INT_R.BYP_ALT1.VCC_WIRE, got CLBLL_R_X21Y7.CLBLL_R.SLICEL_X0.CFFMUX.CX
  00400a9e_015_28: had INT_R_X21Y7.INT_R.BYP_ALT1.VCC_WIRE, got CLBLL_R_X21Y7.CLBLL_R.SLICEL_X0.DFFMUX.CY
  00400a9f_015_05: had INT_R_X21Y7.INT_R.BYP_ALT0.VCC_WIRE, got CLBLL_R_X21Y7.CLBLL_R.SLICEL_X1.CFFMUX.CX
  00400a9f_015_30: had INT_R_X21Y7.INT_R.BYP_ALT0.VCC_WIRE, got CLBLL_R_X21Y7.CLBLL_R.SLICEL_X1.DFFMUX.CY
ERROR: 4 collisions
  0040041e_076_05: had CLBLM_L_X8Y37.CLBLM_L.SLICEM_X0.CFFMUX.CX, got INT_L_X8Y37.INT_L.BYP_ALT1.VCC_WIRE
  0040041e_076_28: had CLBLM_L_X8Y37.CLBLM_L.SLICEM_X0.DFFMUX.CY, got INT_L_X8Y37.INT_L.BYP_ALT1.VCC_WIRE
  0040041f_076_05: had CLBLM_L_X8Y37.CLBLM_L.SLICEL_X1.CFFMUX.CX, got INT_L_X8Y37.INT_L.BYP_ALT0.VCC_WIRE
  0040041f_076_30: had CLBLM_L_X8Y37.CLBLM_L.SLICEL_X1.DFFMUX.CY, got INT_L_X8Y37.INT_L.BYP_ALT0.VCC_WIRE
ERROR: 4 collisions
  0040141e_084_05: had INT_L_X40Y41.INT_L.BYP_ALT1.VCC_WIRE, got CLBLL_L_X40Y41.CLBLL_L.SLICEL_X0.CFFMUX.CX
  0040141e_084_28: had INT_L_X40Y41.INT_L.BYP_ALT1.VCC_WIRE, got CLBLL_L_X40Y41.CLBLL_L.SLICEL_X0.DFFMUX.CY
  0040141f_084_05: had INT_L_X40Y41.INT_L.BYP_ALT0.VCC_WIRE, got CLBLL_L_X40Y41.CLBLL_L.SLICEL_X1.CFFMUX.CX
  0040141f_084_30: had INT_L_X40Y41.INT_L.BYP_ALT0.VCC_WIRE, got CLBLL_L_X40Y41.CLBLL_L.SLICEL_X1.DFFMUX.CY
ERROR: 4 collisions
  0000101e_037_05: had INT_L_X32Y68.INT_L.BYP_ALT1.VCC_WIRE, got CLBLM_L_X32Y68.CLBLM_L.SLICEM_X0.CFFMUX.CX
  0000101e_037_28: had INT_L_X32Y68.INT_L.BYP_ALT1.VCC_WIRE, got CLBLM_L_X32Y68.CLBLM_L.SLICEM_X0.DFFMUX.CY
  0000101f_037_05: had INT_L_X32Y68.INT_L.BYP_ALT0.VCC_WIRE, got CLBLM_L_X32Y68.CLBLM_L.SLICEL_X1.CFFMUX.CX
  0000101f_037_30: had INT_L_X32Y68.INT_L.BYP_ALT0.VCC_WIRE, got CLBLM_L_X32Y68.CLBLM_L.SLICEL_X1.DFFMUX.CY
ERROR: 4 collisions
  00020a1e_043_05: had CLBLM_L_X20Y121.CLBLM_L.SLICEM_X0.CFFMUX.CX, got INT_L_X20Y121.INT_L.BYP_ALT1.VCC_WIRE
  00020a1e_043_28: had CLBLM_L_X20Y121.CLBLM_L.SLICEM_X0.DFFMUX.CY, got INT_L_X20Y121.INT_L.BYP_ALT1.VCC_WIRE
  00020a1f_043_05: had CLBLM_L_X20Y121.CLBLM_L.SLICEL_X1.CFFMUX.CX, got INT_L_X20Y121.INT_L.BYP_ALT0.VCC_WIRE
  00020a1f_043_30: had CLBLM_L_X20Y121.CLBLM_L.SLICEL_X1.DFFMUX.CY, got INT_L_X20Y121.INT_L.BYP_ALT0.VCC_WIRE
ERROR: 4 collisions
  00020f9e_039_05: had CLBLL_R_X31Y119.CLBLL_R.SLICEL_X0.CFFMUX.CX, got INT_R_X31Y119.INT_R.BYP_ALT1.VCC_WIRE
[... cut ...]
  00000181_064_11: had CLBLM_R_X3Y81.CLBLM_R.SLICEM_X0.CLUT.DI1MUX.CI, got INT_R_X3Y81.INT_R.FAN_ALT5.GFAN1
  0000019e_064_05: had CLBLM_R_X3Y81.CLBLM_R.SLICEM_X0.CFFMUX.CX, got INT_R_X3Y81.INT_R.BYP_ALT1.VCC_WIRE
  0000019e_064_28: had CLBLM_R_X3Y81.CLBLM_R.SLICEM_X0.DFFMUX.CY, got INT_R_X3Y81.INT_R.BYP_ALT1.VCC_WIRE
  0000019f_064_05: had CLBLM_R_X3Y81.CLBLM_R.SLICEL_X1.CFFMUX.CX, got INT_R_X3Y81.INT_R.BYP_ALT0.VCC_WIRE
  0000019f_064_30: had CLBLM_R_X3Y81.CLBLM_R.SLICEL_X1.DFFMUX.CY, got INT_R_X3Y81.INT_R.BYP_ALT0.VCC_WIRE
```

mithro, 2019-02-08:
I'm closing this bug as we have individual bugs for current collisions and issues.

closed 2019-02-08 by mithro