f4pga/prjxray #327: Make part files accessible from python

mcmasterg, 2018-12-13:
I tried to load the part files to get frame counts for tilegrid, but I ran into a lot of issues. The main issue seems to be around ! which generates error messages (using import yaml) like "could not determine a constructor for the tag". I tried to add a pass-through constructor, but I just ended up with one giant string. The files also fail to parse in online converters I tried.

Without a current YAML fanbase the preferred option might be to write JSON instead, but C++ tools depend on this file, which I'm not interested in rewritting. So one good option might be to figure out a way to reasonably convert them to .json and publish both.

mcmasterg, 2018-12-13:
Sample file header
```
$ head database/artix7/xc7a50tfgg484-1.yaml
!<xilinx/xc7series/part>
idcode: 0x362c093
global_clock_regions:
  top: !<xilinx/xc7series/global_clock_region>
    rows:
      0: !<xilinx/xc7series/row>
        configuration_buses:
          CLB_IO_CLK: !<xilinx/xc7series/configuration_bus>
            configuration_columns:
              0: !<xilinx/xc7series/configuration_column>
```

mcmasterg, 2018-12-13:
@galv You've poked at the project occasionally. Is this something you'd be interested to look at?

mithro, 2018-12-13:
This seems to do a conversion to JSON,
`ruby -ryaml -rjson -e 'puts JSON.pretty_generate(YAML.load(ARGF))' < xc7a35tcpg236-1.yaml`

kc8apf, 2018-12-13:
If you remove the calls to set the tag and check the tag in the YAML
encoder/decoder methods in libprjxray, the tags should go away. They're not
strictly needed.

If you drop the tags (by modifying the code or manually editing the YAML),
yq should be able to convert it to JSON for you. Hard part is replacing the
C++ YAML library with a JSON library.

commit 70294b33 (2018-12-18): Add wrapper to allow easy loading of yaml files.

Fixes #327.
closed 2018-12-18 by mcmasterg