f4pga/prjxray #377: Add support for Zynq parts

mithro, 2018-12-20:
The Zynq parts have the same fabric as the Artix-7 / Kintex-7. This should make it pretty easy to apply the same fuzzing approach to these parts.

We should add support for them!

mithro, 2018-12-20:
@kgugala has started work on this.

mithro, 2018-12-20:
@kgugala - Can you put the current status here?

mithro, 2018-12-20:
- [ ] Add settings.sh for a Zynq part
 - [ ] Check the status of each of the fuzzers
 - [ ] Create symbiflow-arch-defs for Zynq part

litghost, 2018-12-21:
I recommend fixing https://github.com/SymbiFlow/symbiflow-arch-defs/issues/298 before integrating with symbiflow-arch-defs.  This is with the assumption that the tile types, site types and definitions for zync and artix7 are the same.

mithro, 2019-06-12:
@kgugala This is pretty much done now right?

kgugala, 2019-06-12:
@mithro yes it is. I think we can close this issue and create more specific ones for the Zynq related stuff

closed 2019-06-12 by kgugala