f4pga/prjxray #403: xjson failure

JohnDMcMaster, 2018-12-24:
```
+ python3 generate_grid.py --root_dir specimen_001/ --output_dir output --ignored_wires artix7_ignored_wires.txt
2018-12-24 05:30:41.980389 Running 10 processes
2018-12-24 05:30:42.109208 Reading tilegrid
2018-12-24 05:30:42.237973 Creating tile map
100% (18055 of 18055) |######################################################################################################################################################| Elapsed Time: 0:01:54 Time:  0:01:54
2018-12-24 05:32:51.480714 Reading node tree
2018-12-24 05:33:15.685414 Creating tile connections
100% (1953452 of 1953452) |##################################################################################################################################################| Elapsed Time: 0:04:15 Time:  0:04:15
2018-12-24 05:37:42.705500 Writing tileconn
2018-12-24 05:37:43.956581 Connecting wires to verify tileconn
100% (6193757 of 6193757) |##################################################################################################################################################| Elapsed Time: 0:04:33 Time:  0:04:33
2018-12-24 05:42:51.934957 Verifing tileconn
Traceback (most recent call last):
  File "generate_grid.py", line 676, in <module>
    main()
  File "generate_grid.py", line 656, in main
    xjson.pprint(f, error_nodes)
  File "/mnt/m10_2/buffer/prjxray.2/utils/xjson.py", line 58, in pprint
    sort(data)
  File "/mnt/m10_2/buffer/prjxray.2/utils/xjson.py", line 28, in sort
    o['wire_pairs'].sort(
TypeError: tuple indices must be integers or slices, not str
Makefile:16: recipe for target 'specimen_001/OK' failed
make: *** [specimen_001/OK] Error 1

real    494m1.104s
user    1815m34.500s
sys     249m53.852s
```

mithro, 2018-12-25:
I instead get the following error;
```
100% (150 of 150) |##############################################################################################################################################################################################################################################################################################################################################################################| Elapsed Time: 0:01:30 Time:  0:01:30
+ python3 create_node_tree.py --dump_all_root_dir specimen_001/ --ordered_wires_root_dir ../072-ordered_wires/specimen_001/ --output_dir output
Traceback (most recent call last):
  File "create_node_tree.py", line 279, in <module>
    main()
  File "create_node_tree.py", line 224, in main
    assert os.path.exists(downhill_wires)
AssertionError
Makefile:16: recipe for target 'specimen_001/OK' failed
make: *** [specimen_001/OK] Error 1
```

mithro, 2018-12-25:
Oh, that issue seems to be caused by me not running `072-ordered_wires` first.