f4pga/prjxray #408: 052-clkin fails on Zynq

mithro, 2018-12-27:
```
++ fgrep CRITICAL vivado.log
+ test -z ''
+ for x in 'design*.bit'
+ /var/lib/jenkins/workspace/prjxray-zynq7/build/tools/bitread --part_file /var/lib/jenkins/workspace/prjxray-zynq7/database/zynq7/xc7z010clg400-1.yaml -F 0x00000000:0xffffffff -o design.bits -z -y design.bit
Bitstream size: 2185267 bytes
Config size: 546279 words
Number of configuration frames: 5144
DONE
+ python3 /var/lib/jenkins/workspace/prjxray-zynq7/fuzzers/052-clkin/generate.py
../int_loop.mk:59: recipe for target 'build/50/specimen_007/OK' failed
make[2]: *** [build/50/specimen_007/OK] Error 1
Loading bits from design.bits.
Loading tags from design.txt.
Compiling segment data.
Traceback (most recent call last):
  File "/var/lib/jenkins/workspace/prjxray-zynq7/fuzzers/052-clkin/generate.py", line 66, in <module>
    segmk.compile(bitfilter=bitfilter)
  File "/var/lib/jenkins/workspace/prjxray-zynq7/prjxray/segmaker.py", line 360, in compile
    add_tilename_tags()
  File "/var/lib/jenkins/workspace/prjxray-zynq7/prjxray/segmaker.py", line 256, in add_tilename_tags
    segment = getseg(segname)
  File "/var/lib/jenkins/workspace/prjxray-zynq7/prjxray/segmaker.py", line 245, in getseg
    segments, segname, tiledata, bitfilter=bitfilter)
  File "/var/lib/jenkins/workspace/prjxray-zynq7/prjxray/segmaker.py", line 231, in add_segbits
    bitname_bit):
  File "/var/lib/jenkins/workspace/prjxray-zynq7/fuzzers/052-clkin/generate.py", line 58, in bitfilter
    assert os.getenv("XRAY_DATABASE") in ["artix7", "kintex7"]
AssertionError
```

JohnDMcMaster, 2018-12-27:
Suggest we remove all of these asserts that are not truly part specific (ie ones that are ["artix7", "kintex7"] as opposed to just "artix7"). This should be all of the fuzzers, although some minitests really are part specific

acomodi, 2019-01-02:
I have temporarily added "zynq7" to the list so not to end up in assertion errors for zynq parts.
Is it better to remove all these asserts from the fuzzers (they should be fuzzer 051, 052, 053 and 057) then or we keep them?

litghost, 2019-01-17:
Is this fixed?

acomodi, 2019-01-29:
Fixed

closed 2019-01-29 by kgugala