f4pga/prjxray #417: 005-tilegrid fails with Python error

GuzTech, 2018-12-28:
When running all fuzzers, I get the following error:

```+ '[' -f /home/oguz286/Projects/FPGA/prjxray/fuzzers/005-tilegrid/iob/top.py ']'
+ XRAY_DATABASE_ROOT=/home/oguz286/Projects/FPGA/prjxray/fuzzers/005-tilegrid/iob/../build/basicdb
+ python3 /home/oguz286/Projects/FPGA/prjxray/fuzzers/005-tilegrid/iob/top.py
Traceback (most recent call last):
  File "/home/oguz286/Projects/FPGA/prjxray/fuzzers/005-tilegrid/iob/top.py", line 148, in <module>
    run()
  File "/home/oguz286/Projects/FPGA/prjxray/fuzzers/005-tilegrid/iob/top.py", line 70, in run
    assign_i(rand_site(), 'di[0]')
  File "/home/oguz286/Projects/FPGA/prjxray/fuzzers/005-tilegrid/iob/top.py", line 51, in rand_site
    return random.choice(list(remain_sites()))
  File "/usr/lib/python3.7/random.py", line 261, in choice
    raise IndexError('Cannot choose from an empty sequence') from None
IndexError: Cannot choose from an empty sequence
```
This indicates that `remain_sites()` returns an empty set.

JohnDMcMaster, 2018-12-28:
Which part is this? artix7 or something else?

GuzTech, 2018-12-28:
This is for the artix7.

JohnDMcMaster, 2018-12-29:
Sounds like there are no IOB33S's in your region of interest. Are you using Vivado 2017.2? Can you send your tilegrid.json?

Are you in freenode IRC ##symbiflow by chance? May make debugging this easier

GuzTech, 2018-12-29:
Yes, I'm using 2017.2. The file is too big to paste on paste.gnome.org for example. I did find the string `IOB33S` in the file 96 times though.

I'm in #symbiflow now as GuzTech.

closed 2018-12-31 by GuzTech