f4pga/prjxray #418: 051-imuxlout slow builds / timeout

JohnDMcMaster, 2018-12-29:
It doesn't appear stuck, but it is timing out after 50 runs or so. I think this is related to recent -m/-M changes, but its effecting things much more strongly than I was expecting.

TODO file sizes on prjxray-bit-artix7/6
```
	1_all.txt	7.70 KB	view
	2_all.txt	7.70 KB	view
	3_all.txt	7.70 KB	view
	4_all.txt	7.70 KB	view
	5_all.txt	7.70 KB	view
	6_all.txt	7.70 KB	view
	7_all.txt	7.70 KB	view
	8_all.txt	7.70 KB	view
	9_all.txt	7.70 KB	view
	10_all.txt	7.70 KB	view
	11_all.txt	7.64 KB	view
	12_all.txt	7.64 KB	view
	13_all.txt	7.64 KB	view
	14_all.txt	7.64 KB	view
	15_all.txt	7.62 KB	view
	16_all.txt	7.59 KB	view
	17_all.txt	7.59 KB	view
	18_all.txt	7.59 KB	view
	19_all.txt	7.59 KB	view
	20_all.txt	7.56 KB	view
	21_all.txt	7.51 KB	view
	22_all.txt	7.51 KB	view
	23_all.txt	7.51 KB	view
	24_all.txt	7.48 KB	view
	25_all.txt	7.48 KB	view
	26_all.txt	7.48 KB	view
	27_all.txt	7.48 KB	view
	28_all.txt	7.48 KB	view
	29_all.txt	7.48 KB	view
	30_all.txt	7.48 KB	view
	31_all.txt	7.48 KB	view
	32_all.txt	7.48 KB	view
	33_all.txt	7.48 KB	view
	34_all.txt	7.45 KB	view
	35_all.txt	7.45 KB	view
	36_all.txt	7.42 KB	view
	37_all.txt	7.42 KB	view
	38_all.txt	7.42 KB	view
	39_all.txt	7.42 KB	view
	40_all.txt	7.40 KB	view
	41_all.txt	7.37 KB	view
	42_all.txt	7.37 KB	view
	43_all.txt	7.37 KB	view
	44_all.txt	7.37 KB	view
	45_all.txt	7.37 KB	view
	46_all.txt	7.34 KB	view
	47_all.txt	7.34 KB	view
	48_all.txt	7.34 KB	view
	49_all.txt	7.34 KB	view
	50_all.txt	7.32 KB	view
	51_all.txt	7.32 KB	view
```

mithro, 2019-01-07:
I just hit this when building,
```
+ make ITER=51 cleaniter
make[2]: Entering directory '/big/github/SymbiFlow/prjxray/fuzzers/051-imuxlout'
rm -rf build/51 build/todo.txt
make[2]: Leaving directory '/big/github/SymbiFlow/prjxray/fuzzers/051-imuxlout'
+ make ITER=51 build/todo.txt
make[2]: Entering directory '/big/github/SymbiFlow/prjxray/fuzzers/051-imuxlout'
XRAY_DATABASE_DIR=/big/github/SymbiFlow/prjxray/fuzzers/051-imuxlout/build/database python3 /big/github/SymbiFlow/prjxray/fuzzers/int_maketodo.py --pip-type pips_int --re "^INT_[LR].IMUX(_L)?[0-9]+\.LOGIC_OUTS(_L)?[0-9]+$" |sort >build/todo_all.txt
cat build/todo_all.txt | sort -R | head -n10 > build/todo.txt.tmp
mv build/todo.txt.tmp build/todo.txt
# Per iter files
mkdir -p build/51
cp build/todo_all.txt build/todo.txt build/51/
# All in one dir for easier trending
mkdir -p build/todo
cp build/todo_all.txt build/todo/51_all.txt
make[2]: Leaving directory '/big/github/SymbiFlow/prjxray/fuzzers/051-imuxlout'
+ python3 /big/github/SymbiFlow/prjxray/fuzzers/int_loop_check.py --zero-entries --timeout-iters 50
   1 f505eda3    288 lines      1 stable
   2 f505eda3    288 lines      2 stable
   3 f505eda3    288 lines      3 stable
   4 f505eda3    288 lines      4 stable
   5 f505eda3    288 lines      5 stable
   6 f505eda3    288 lines      6 stable
   7 f505eda3    288 lines      7 stable
   8 f505eda3    288 lines      8 stable
   9 f505eda3    288 lines      9 stable
  10 f505eda3    288 lines     10 stable
  11 f505eda3    288 lines     11 stable
  12 f505eda3    288 lines     12 stable
  13 f505eda3    288 lines     13 stable
  14 f505eda3    288 lines     14 stable
  15 f505eda3    288 lines     15 stable
  16 f505eda3    288 lines     16 stable
  17 c3924886    285 lines      1 stable
  18 c3924886    285 lines      2 stable
  19 c2c072e1    283 lines      1 stable
  20 c2c072e1    283 lines      2 stable
  21 c2c072e1    283 lines      3 stable
  22 c2c072e1    283 lines      4 stable
  23 51aacd0e    282 lines      1 stable
  24 51aacd0e    282 lines      2 stable
  25 51aacd0e    282 lines      3 stable
  26 51aacd0e    282 lines      4 stable
  27 51aacd0e    282 lines      5 stable
  28 51aacd0e    282 lines      6 stable
  29 64559241    280 lines      1 stable
  30 9443ec46    279 lines      1 stable
  31 9443ec46    279 lines      2 stable
  32 9443ec46    279 lines      3 stable
  33 7e4b6eba    278 lines      1 stable
  34 7e4b6eba    278 lines      2 stable
  35 9d67416c    277 lines      1 stable
  36 9d67416c    277 lines      2 stable
  37 9d67416c    277 lines      3 stable
  38 9d67416c    277 lines      4 stable
  39 9d67416c    277 lines      5 stable
  40 9d67416c    277 lines      6 stable
  41 923e73fe    276 lines      1 stable
  42 923e73fe    276 lines      2 stable
  43 923e73fe    276 lines      3 stable
  44 923e73fe    276 lines      4 stable
  45 65a1d0b8    275 lines      1 stable
  46 65a1d0b8    275 lines      2 stable
  47 72172d9d    273 lines      1 stable
  48 72172d9d    273 lines      2 stable
  49 72172d9d    273 lines      3 stable
  50 72172d9d    273 lines      4 stable
  51 72172d9d    273 lines      5 stable
ERROR: timeout (max 50, got 51)
+ '[' -f build/todo/timeout ']'
+ echo 'ERROR: timeout'
ERROR: timeout
+ exit 1
../int_loop.mk:88: recipe for target 'run' failed
make[1]: *** [run] Error 1
make[1]: Leaving directory '/big/github/SymbiFlow/prjxray/fuzzers/051-imuxlout'
Makefile:37: recipe for target '051-imuxlout/run.ok' failed
make: *** [051-imuxlout/run.ok] Error 2
```

```
commit 5fe439765dcadb664527a2a8a3f319e39c7e5529
Merge: 4478859 1b48dfd
Author: John Mcmaster <JohnDMcMaster@gmail.com>
Date:   Thu Dec 27 03:22:53 2018 -0800

    Merge pull request #409 from mithro/master

    Fixes #403.
```

mcmasterg, 2019-01-09:
Some progress on this. Sounds like the general idea is that the code encourages using these routes, but Vivado often chooses not to take them. I'm guessing we made some small change (perhaps related to ROI exclusion?) that biased the decision. @kgugala is looking at using route_via to improve this. I've also suggested looking at Clifford's original fuzzer to see if there might be some sort of major regression there

mithro, 2019-01-09:
This might be related to https://github.com/SymbiFlow/prjxray/issues/472

mcmasterg, 2019-01-10:
this PR is supposed to give a significant improvement: https://github.com/SymbiFlow/prjxray/pull/474

Closing for now. Reopen if continue to have problems

closed 2019-01-10 by mcmasterg