f4pga/prjxray #432: Fuzzer 017-clbprecyinit does not generate PRECYINIT.0 bit

kgugala, 2019-01-02:
when handling fasm with `CLBLL_L_X16Y50.SLICEL_X1.PRECYINIT.0` the fasm2frames fails with `Expected Identifier at position ...`

This is because `...PRECYINIT.0` could not be find in the database (017 fuzzer does not generate it)

mithro, 2019-01-02:
@JohnDMcMaster My guess here was that it is because the `CLBLL_L_X16Y50.SLICEL_X1.PRECYINIT.0` is effectively used when *no* other option for the MUX is selected. How is that represented in FASM & the database file?

JohnDMcMaster, 2019-01-02:
There is supposed to be a mask correction dbr to .db file

kgugala, 2019-01-02:
the problem is not related to fuzzer 017, but to fasm assembler. I do not allow the identifiers to  start with a number. I have opened a PR fixing that in fasm https://github.com/SymbiFlow/fasm/pull/2

I'm closing this one

closed 2019-01-02 by kgugala