f4pga/prjxray #439: 005-tilegrid fails for Zynq

Fatsie, 2019-01-03:
I set up  prjxray according to documentation. I am using Vivado 2018.2 though.
Running make in fuzzers failed in 005-tilegrid/iob; [vivado.log](https://github.com/SymbiFlow/prjxray/files/2725222/vivado.log) from 005-tilegrid/iob/build/specimen_001/

mithro, 2019-01-04:
Hi @Fatsie,

At the moment prjxray only works with Vivado 2017.02 - newer versions of Vivado introduced error checking which causes a bunch of the fuzzers to fail.

Can you try installing the correct version of Vivado and give it another try?

I should also mentioned that Zynq support was only started a couple of weeks ago, so it is still *very* new.

Looking for your update!

Tim '@mithro' Ansell

closed 2019-01-04 by JohnDMcMaster

Fatsie, 2019-01-04:
In the mean time I installed 2017.2 and can confirm full build is now running OK; already past 005-tilegrid.
Did have to do a make clean first otherwise he still failed.

JohnDMcMaster, 2019-01-04:
Fyi current -m options on 05x series as is now may cause run to take a long time