f4pga/prjxray #440: Broken VCC_WIRE entries in database

kgugala, 2019-01-04:
see https://github.com/SymbiFlow/symbiflow-arch-defs/pull/314#issuecomment-451496422

mcmasterg, 2019-01-07:
From quick discussion sounds like we never had a known good entry. Oddly there is a GND fuzzer and a VCC fuzzer. I'm guessing that these simply aren't used a lot and so hasn't been a problem until recently?

litghost, 2019-01-07:
In the previous database, VCC entries were located in ppips as "default". See https://github.com/SymbiFlow/prjxray-db/commit/b327fb4b53299cd7a11742b95181321b2d398074#diff-5693698103ebe6ceff1d3b92b9a7fedb

Then in the Oct 21st update, these entries were removed: https://github.com/SymbiFlow/prjxray-db/commit/fd147ad56902269851a3e6382f8f4edd39a163a4#diff-5693698103ebe6ceff1d3b92b9a7fedb

mcmasterg, 2019-01-07:
I would believe that. I didn't have a good understand of these when I was refactoring things. Probably should get a code review on how these are implemented in the fuzzers. FWIW I think someone is consolidating pip fuzzer .py right now

closed 2019-01-28 by litghost