f4pga/prjxray #444: tilegrid: automatically generate number of frames

mcmasterg, 2019-01-07:
Currently tilegrid code hard codes the number of frames a tile uses. However, this information is known after partgen in run. We should automatically read this from the json or yaml file.

The following locations have hard coded entries:
- add_tdb.py: see tdb_fns
- generate_full.py:db_add_bits(): see "entry = {"

Prioritize add_tdb.py as the code in generate_full.py might get phased out in favor of add_tdb.py.

For example, this tilegrid entry:
```
    "CLBLL_L_X12Y135": {
        "bits": {
            "CLB_IO_CLK": {
                "baseaddr": "0x00020600",
                "frames": 36,
                "height": 2,
                "offset": 71,
                "words": 2
            }
```

Roughly corresponds to this information in the part .json (the 36 part):

```
  "global_clock_regions": {
    "top": {
      "rows": {
        "0": {
          "configuration_buses": {
            "CLB_IO_CLK": {
              "configuration_columns": {
               ...
                "2": {
                  "frame_count": 36
                },
                "3": {
                  "frame_count": 36
                },
``` 

For details on how to do this transformation, see "Frame Address Register" on page 110: https://www.xilinx.com/support/documentation/user_guides/ug470_7Series_Config.pdf

mithro, 2019-02-08:
@litghost Any thoughts here?

mithro, 2019-06-12:
@litghost What is the status of this?